emlearn/emlearn #1: error while installing

armando-fandango, 2018-10-18:
Getting the following errir while installing on MacBook:

pip install emtrees
Collecting emtrees
  Using cached https://files.pythonhosted.org/packages/79/a1/4fe14ed9dfde36cb9e5796b80c5a12e8f4ad73940113fc3785bbb15da7ec/emtrees-0.2.3.tar.gz
    Complete output from command python setup.py egg_info:
    Traceback (most recent call last):
      File "<string>", line 1, in <module>
      File "/private/var/folders/zq/gskkvxy51kscx6ld0jwlrtwc0000gn/T/pip-install-tv_81i_p/emtrees/setup.py", line 117, in <module>
        install_requires=read_requirements(),
      File "/private/var/folders/zq/gskkvxy51kscx6ld0jwlrtwc0000gn/T/pip-install-tv_81i_p/emtrees/setup.py", line 93, in read_requirements
        with open(requirements_txt, encoding='utf-8') as f:
    FileNotFoundError: [Errno 2] No such file or directory: '/private/var/folders/zq/gskkvxy51kscx6ld0jwlrtwc0000gn/T/pip-install-tv_81i_p/emtrees/requirements.txt'

armando-fandango, 2018-10-18:
I just tested from master branch and it works:
`python3 -m pip install --index-url https://test.pypi.org/simple/ emtrees`

Hence probably just the dist is needed to be updated to pypi

jonnor, 2018-10-20:
Hi @armando-fandango , thank you for reporting! Some files were missing in the distribution bundle, should be fixed in `0.2.4` which is on PyPi in a few minutes.

PS: There will be some breaking changes going into master soon. Am planning to merge some of my other embedded machine learning packages (Gaussian Naive Bayes, ANN, audio feature extraction) and rename to `emlearn`.

commit f63315d1 (2018-10-20): Add missing files in Python package

Fixes #1
closed 2018-10-20 by jonnor