emlearn/emlearn #2: Cannot install emlearn (VS cl error)

skjerns, 2018-11-02:
Just stumbled up this library and it seems very useful.

However, I receive an error when installing

```
Collecting emlearn
  Using cached https://files.pythonhosted.org/packages/38/53/c61639dd4f20f47e9755bb9848dccd62bfaaa86
1d35ad28224bd496c522d/emlearn-0.4.0.tar.gz
Requirement already satisfied: pybind11>=2.2 in c:\anaconda3\lib\site-packages (from emlearn) (2.2.4
)
Requirement already satisfied: numpy>=1.14.0 in c:\anaconda3\lib\site-packages (from emlearn) (1.15.
1)
Building wheels for collected packages: emlearn
  Running setup.py bdist_wheel for emlearn ... error
  Complete output from command c:\anaconda3\python.exe -u -c "import setuptools, tokenize;__file__='
C:\\Users\\usernameK~1\\AppData\\Local\\Temp\\pip-install-q9yatq5x\\emlearn\\setup.py';f=getattr(tokeni
ze, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file_
_, 'exec'))" bdist_wheel -d C:\Users\usernameK~1\AppData\Local\Temp\pip-wheel-a2crcrq6 --python-tag cp3
7:
  running bdist_wheel
  running build
  running build_py
  creating build
  creating build\lib.win-amd64-3.7
  creating build\lib.win-amd64-3.7\emlearn
  copying emlearn\bayes.py -> build\lib.win-amd64-3.7\emlearn
  copying emlearn\common.py -> build\lib.win-amd64-3.7\emlearn
  copying emlearn\convert.py -> build\lib.win-amd64-3.7\emlearn
  copying emlearn\net.py -> build\lib.win-amd64-3.7\emlearn
  copying emlearn\trees.py -> build\lib.win-amd64-3.7\emlearn
  copying emlearn\__init__.py -> build\lib.win-amd64-3.7\emlearn
  running egg_info
  writing emlearn.egg-info\PKG-INFO
  writing dependency_links to emlearn.egg-info\dependency_links.txt
  writing requirements to emlearn.egg-info\requires.txt
  writing top-level names to emlearn.egg-info\top_level.txt
  reading manifest file 'emlearn.egg-info\SOURCES.txt'
  reading manifest template 'MANIFEST.in'
  writing manifest file 'emlearn.egg-info\SOURCES.txt'
  copying emlearn\eml_audio.h -> build\lib.win-amd64-3.7\emlearn
  copying emlearn\eml_bayes.h -> build\lib.win-amd64-3.7\emlearn
  copying emlearn\eml_benchmark.h -> build\lib.win-amd64-3.7\emlearn
  copying emlearn\eml_common.h -> build\lib.win-amd64-3.7\emlearn
  copying emlearn\eml_fft.h -> build\lib.win-amd64-3.7\emlearn
  copying emlearn\eml_fixedpoint.h -> build\lib.win-amd64-3.7\emlearn
  copying emlearn\eml_net.h -> build\lib.win-amd64-3.7\emlearn
  copying emlearn\eml_test.h -> build\lib.win-amd64-3.7\emlearn
  copying emlearn\eml_trees.h -> build\lib.win-amd64-3.7\emlearn
  copying emlearn\eml_vector.h -> build\lib.win-amd64-3.7\emlearn
  running build_ext
  building 'eml_trees' extension
  creating build\temp.win-amd64-3.7
  creating build\temp.win-amd64-3.7\Release
  creating build\temp.win-amd64-3.7\Release\bindings
  C:\Program Files (x86)\Microsoft Visual Studio 14.0\VC\BIN\x86_amd64\cl.exe /c /nologo /Ox /W3 /GL
 /DNDEBUG /MD -Iemlearn/ -Ic:\anaconda3\Include -IC:\Users\usernameAppData\Roaming\Python\Python37
\Include -Ic:\anaconda3\include -Ic:\anaconda3\include "-IC:\Program Files (x86)\Microsoft Visual St
udio 14.0\VC\INCLUDE" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.17763.0\ucrt" "-IC:\Pro
gram Files (x86)\Windows Kits\NETFXSDK\4.6\include\um" "-IC:\Program Files (x86)\Windows Kits\10\inc
lude\10.0.17763.0\shared" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.17763.0\um" "-IC:\P
rogram Files (x86)\Windows Kits\10\include\10.0.17763.0\winrt" /EHsc /Tpbindings/eml_trees.cpp /Fobu
ild\temp.win-amd64-3.7\Release\bindings/eml_trees.obj /EHsc /DVERSION_INFO=\"0.4.0\"
  eml_trees.cpp
  bindings/eml_trees.cpp(24): warning C4267: 'initializing': conversion from 'size_t' to 'int', poss
ible loss of data
  bindings/eml_trees.cpp(24): warning C4267: 'initializing': conversion from 'size_t' to 'const int'
, possible loss of data
  bindings/eml_trees.cpp(39): warning C4267: '=': conversion from 'size_t' to 'int32_t', possible lo
ss of data
  bindings/eml_trees.cpp(54): warning C4244: 'initializing': conversion from 'const pybind11::ssize_
[... 235 lines cut ...]
    bindings/eml_audio.cpp(61): note: failure was caused by non-constant arguments or reference to a
 non-constant symbol
    bindings/eml_audio.cpp(61): note: see usage of 'length'
    bindings/eml_audio.cpp(66): error C4576: a parenthesized type followed by an initializer list is
 a non-standard explicit type conversion syntax
    bindings/eml_audio.cpp(81): warning C4996: 'sscanf': This function or variable may be unsafe. Co
nsider using sscanf_s instead. To disable deprecation, use _CRT_SECURE_NO_WARNINGS. See online help
for details.
    C:\Program Files (x86)\Windows Kits\10\include\10.0.17763.0\ucrt\stdio.h(2263): note: see declar
ation of 'sscanf'
    error: command 'C:\\Program Files (x86)\\Microsoft Visual Studio 14.0\\VC\\BIN\\x86_amd64\\cl.ex
e' failed with exit status 2

    ----------------------------------------
Command "c:\anaconda3\python.exe -u -c "import setuptools, tokenize;__file__='C:\\Users\\usernameK~1\\A
ppData\\Local\\Temp\\pip-install-q9yatq5x\\emlearn\\setup.py';f=getattr(tokenize, 'open', open)(__fi
le__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" install -
-record C:\Users\usernameK~1\AppData\Local\Temp\pip-record-dxwqa6qd\install-record.txt --single-version
-externally-managed --compile --user --prefix=" failed with error code 1 in C:\Users\usernameK~1\AppDat
a\Local\Temp\pip-install-q9yatq5x\emlearn\
```

do you know anything about that? Are there any wheels available?

I'm running Windows 10 x64.

jonnor, 2018-11-12:
Hi @skjerns and thanks for reporting! It looks like there are some (minor) compile issues with the Python bindings on Windows. Have not tested on Windows (or OSX) yet, but I'll activate Travis CI for those now and see if we can fix these compile errors.

jonnor, 2018-11-12:
Had to use Appveyer instead of Travis CI, but the package compiles now at least. https://github.com/jonnor/emlearn/pull/3

jonnor, 2018-11-13:
Package builds and tests pass on Windows now. Released as `emlearn 0.4.1`, which looks to install OK.

No binary wheels at this time, so still need to have a C++ compiler for the bindings.

skjerns, 2018-11-21:
it works now, thanks for the quick response!!