emlearn/emlearn #7: CL install error on 0.8.0

skjerns, 2019-03-06:
Hey!

Just wanted to install the newest version, however I get

` Cannot open include file: 'emlpy_common.hpp': No such file or directory`

Full log:

<details><summary>Full log output </summary>
<p>

```
```
Collecting emlearn
  Using cached https://files.pythonhosted.org/packages/10/7a/bc4f241a34dfc608d1ec1259b8fe5cc5d588e58a43be06e2475cf66863c3/emlearn-0.8.0.tar.gz
Requirement already satisfied, skipping upgrade: pybind11>=2.2 in c:\users\UserName\appdata\roaming\python\python36\site-packages (from emlearn) (2.2.4)
Requirement already satisfied, skipping upgrade: numpy>=1.14.0 in c:\anaconda3\lib\site-packages (from emlearn) (1.15.4)
Building wheels for collected packages: emlearn
  Building wheel for emlearn (setup.py) ... error
  Complete output from command c:\anaconda3\python.exe -u -c "import setuptools, tokenize;__file__='C:\\Users\\UserName\\AppData\\Local\\Temp\\pip-install-koeyjk7s\\emlearn\\setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" bdist_wheel -d C:\Users\UserName\AppData\Local\Temp\pip-wheel-i7yw98rb --python-tag cp36:
  c:\anaconda3\lib\distutils\dist.py:261: UserWarning: Unknown distribution option: 'long_description_content_type'
    warnings.warn(msg)
  running bdist_wheel
  running build
  running build_py
  creating build
  creating build\lib.win-amd64-3.6
  creating build\lib.win-amd64-3.6\emlearn
  copying emlearn\bayes.py -> build\lib.win-amd64-3.6\emlearn
  copying emlearn\cgen.py -> build\lib.win-amd64-3.6\emlearn
  copying emlearn\common.py -> build\lib.win-amd64-3.6\emlearn
  copying emlearn\convert.py -> build\lib.win-amd64-3.6\emlearn
  copying emlearn\net.py -> build\lib.win-amd64-3.6\emlearn
  copying emlearn\signal.py -> build\lib.win-amd64-3.6\emlearn
  copying emlearn\trees.py -> build\lib.win-amd64-3.6\emlearn
  copying emlearn\__init__.py -> build\lib.win-amd64-3.6\emlearn
  running egg_info
  writing emlearn.egg-info\PKG-INFO
  writing dependency_links to emlearn.egg-info\dependency_links.txt
  writing requirements to emlearn.egg-info\requires.txt
  writing top-level names to emlearn.egg-info\top_level.txt
  reading manifest file 'emlearn.egg-info\SOURCES.txt'
  reading manifest template 'MANIFEST.in'
  writing manifest file 'emlearn.egg-info\SOURCES.txt'
  copying emlearn\eml_audio.h -> build\lib.win-amd64-3.6\emlearn
  copying emlearn\eml_bayes.h -> build\lib.win-amd64-3.6\emlearn
  copying emlearn\eml_benchmark.h -> build\lib.win-amd64-3.6\emlearn
  copying emlearn\eml_common.h -> build\lib.win-amd64-3.6\emlearn
  copying emlearn\eml_fft.h -> build\lib.win-amd64-3.6\emlearn
  copying emlearn\eml_fixedpoint.h -> build\lib.win-amd64-3.6\emlearn
  copying emlearn\eml_iir.h -> build\lib.win-amd64-3.6\emlearn
  copying emlearn\eml_net.h -> build\lib.win-amd64-3.6\emlearn
  copying emlearn\eml_test.h -> build\lib.win-amd64-3.6\emlearn
  copying emlearn\eml_trees.h -> build\lib.win-amd64-3.6\emlearn
  copying emlearn\eml_vector.h -> build\lib.win-amd64-3.6\emlearn
  running build_ext
  building 'eml_trees' extension
  creating build\temp.win-amd64-3.6
  creating build\temp.win-amd64-3.6\Release
  creating build\temp.win-amd64-3.6\Release\bindings
  C:\Program Files (x86)\Microsoft Visual Studio\2017\Community\VC\Tools\MSVC\14.15.26726\bin\HostX86\x64\cl.exe /c /nologo /Ox /W3 /GL /DNDEBUG /MD -Iemlearn/ -Ic:\anaconda3\Include -IC:\Users\UserName\AppData\Roaming\Python\Python36\Include -Ic:\anaconda3\include -Ic:\anaconda3\include "-IC:\Program Files (x86)\Microsoft Visual Studio\2017\Community\VC\Tools\MSVC\14.15.26726\ATLMFC\include" "-IC:\Program Files (x86)\Microsoft Visual Studio\2017\Community\VC\Tools\MSVC\14.15.26726\include" "-IC:\Program Files (x86)\Windows Kits\NETFXSDK\4.6\include\um" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.17763.0\ucrt" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.17763.0\shared" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.17763.0\um" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.17763.0\winrt" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.17763.0\cppwinrt" /EHsc /Tpbindings/eml_trees.cpp /Fobuild\temp.win-amd64-3.6\Release\bindings/eml_trees.obj /EHsc /DVERSION_INFO=\"0.8.0\"
  eml_trees.cpp
  bindings/eml_trees.cpp(24): warning C4267: 'initializing': conversion from 'size_t' to 'int', possible loss of data
  bindings/eml_trees.cpp(24): warning C4267: 'initializing': conversion from 'size_t' to 'const int', possible loss of data
  bindings/eml_trees.cpp(39): warning C4267: '=': conversion from 'size_t' to 'int32_t', possible loss of data
  bindings/eml_trees.cpp(54): warning C4244: 'initializing': conversion from 'const pybind11::ssize_t' to 'int32_t', possible loss of data
  bindings/eml_trees.cpp(54): warning C4244: 'initializing': conversion from 'const pybind11::ssize_t' to 'const int32_t', possible loss of data
  bindings/eml_trees.cpp(75): warning C4996: 'sscanf': This function or variable may be unsafe. Consider using sscanf_s instead. To disable deprecation, use _CRT_SECURE_NO_WARNINGS. See online help for details.
  C:\Program Files (x86)\Windows Kits\10\include\10.0.17763.0\ucrt\stdio.h(2263): note: see declaration of 'sscanf'
  C:\Program Files (x86)\Microsoft Visual Studio\2017\Community\VC\Tools\MSVC\14.15.26726\bin\HostX86\x64\link.exe /nologo /INCREMENTAL:NO /LTCG /DLL /MANIFEST:EMBED,ID=2 /MANIFESTUAC:NO /LIBPATH:c:\anaconda3\libs /LIBPATH:c:\anaconda3\PCbuild\amd64 "/LIBPATH:C:\Program Files (x86)\Microsoft Visual Studio\2017\Community\VC\Tools\MSVC\14.15.26726\ATLMFC\lib\x64" "/LIBPATH:C:\Program Files (x86)\Microsoft Visual Studio\2017\Community\VC\Tools\MSVC\14.15.26726\lib\x64" "/LIBPATH:C:\Program Files (x86)\Windows Kits\NETFXSDK\4.6\lib\um\x64" "/LIBPATH:C:\Program Files (x86)\Windows Kits\10\lib\10.0.17763.0\ucrt\x64" "/LIBPATH:C:\Program Files (x86)\Windows Kits\10\lib\10.0.17763.0\um\x64" /EXPORT:PyInit_eml_trees build\temp.win-amd64-3.6\Release\bindings/eml_trees.obj /OUT:build\lib.win-amd64-3.6\eml_trees.cp36-win_amd64.pyd /IMPLIB:build\temp.win-amd64-3.6\Release\bindings\eml_trees.cp36-win_amd64.lib
[... 99 lines cut ...]
    emlearn/eml_audio.h(106): warning C4244: 'initializing': conversion from 'float' to 'int', possible loss of data
    emlearn/eml_audio.h(106): warning C4244: 'initializing': conversion from 'float' to 'const int', possible loss of data
    bindings/eml_audio.cpp(14): fatal error C1083: Cannot open include file: 'emlpy_common.hpp': No such file or directory
    error: command 'C:\\Program Files (x86)\\Microsoft Visual Studio\\2017\\Community\\VC\\Tools\\MSVC\\14.15.26726\\bin\\HostX86\\x64\\cl.exe' failed with exit status 2

    ----------------------------------------
  Rolling back uninstall of emlearn
  Moving to c:\users\UserName\appdata\roaming\python\python36\site-packages\eml_audio.cp36-win_amd64.pyd
   from C:\Users\UserName\AppData\Local\Temp\pip-uninstall-jvnmg8kz\eml_audio.cp36-win_amd64.pyd
  Moving to c:\users\UserName\appdata\roaming\python\python36\site-packages\eml_bayes.cp36-win_amd64.pyd
   from C:\Users\UserName\AppData\Local\Temp\pip-uninstall-jvnmg8kz\eml_bayes.cp36-win_amd64.pyd
  Moving to c:\users\UserName\appdata\roaming\python\python36\site-packages\eml_net.cp36-win_amd64.pyd
   from C:\Users\UserName\AppData\Local\Temp\pip-uninstall-jvnmg8kz\eml_net.cp36-win_amd64.pyd
  Moving to c:\users\UserName\appdata\roaming\python\python36\site-packages\eml_trees.cp36-win_amd64.pyd
   from C:\Users\UserName\AppData\Local\Temp\pip-uninstall-jvnmg8kz\eml_trees.cp36-win_amd64.pyd
  Moving to c:\users\UserName\appdata\roaming\python\python36\site-packages\emlearn-0.5.1-py3.6.egg-info
   from c:\users\UserName\appdata\roaming\python\python36\site-packages\~mlearn-0.5.1-py3.6.egg-info
  Moving to c:\users\UserName\appdata\roaming\python\python36\site-packages\emlearn\
   from c:\users\UserName\appdata\roaming\python\python36\site-packages\~mlearn
Command "c:\anaconda3\python.exe -u -c "import setuptools, tokenize;__file__='C:\\Users\\UserName\\AppData\\Local\\Temp\\pip-install-koeyjk7s\\emlearn\\setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" install --record C:\Users\UserName\AppData\Local\Temp\pip-record-t8w6saws\install-record.txt --single-version-externally-managed --compile" failed with error code 1 in C:\Users\UserName\AppData\Local\Temp\pip-install-koeyjk7s\emlearn\
```
```

</p>
</details>

jonnor, 2019-03-06:
Hi again! :D Thank you for reporting.

The file was indeed missing from the Python source packages. This has now been fixed in emlearn 0.8.1. I need to start using emlearn from PyPI instead of git to catch these things...

I also fixed some Linux-isms in the tests, so now we are back to green on Appveyor/Windows again.

closed 2019-03-06 by jonnor

skjerns, 2019-03-07:
great! thank's for the quick fix. Now I can get going exporting my DecisionTrees :))