emlearn/emlearn #8: Depencendy on pybind11 / C99 Code

skjerns, 2019-05-14:
I'm trying to convert to pure C code, without any C++.

As far as I can see `emlearn` created C code uses some dependency on `pybind11` which is for C++.

The `eml_trees.h` declares some classes which I cannot use on my C platform.

Is there an option I'm missing? I'm just following the example.
The `method=` option does not seem to do much?

jonnor, 2019-05-14:
What is is the error you are getting? What compiler is used, and how did you generate the C code?

The pybind11 and C++ code is only supposed to be for the Python bindings. However it is the main target exercising the code right now, and I've mostly also been testing on Arduino (C++ compiler) - so its possible some C++isms have snuck in by accident.

skjerns, 2019-05-15:
I tried to generate C code, as it was written in the description that `emlearn` would be able to generate C99 code.

```
import numpy as np
import emlearn
from sklearn.ensemble import RandomForestClassifier

train_x = np.random.rand(10000, 8)
train_y = np.random.randint(0, 4, 10000)

rfc = RandomForestClassifier(n_estimators=10, max_depth=10)
rfc.fit(train_x, train_y)

# transfer code
cmodel = emlearn.convert(rfc, method='inline')
a=cmodel.save(file='rfc.h')
```

The resulting file `rfc.h` contains an `# include <eml_tree.h>` which contains C++ code, and also uses `EmlTreesNode` objects, which are C++.

Or did I misunderstand something?

jonnor, 2019-05-16:
eml_tree.h or EmlTreesNode are not C++, just C99. It should compiles fine with C99. See example below using GCC 8.3.0. If you get some compile error with your C99 compiler, please provide your compiler name and version, the error being produced and upload the code being compiled.

Both the specialized inline code and the general code using a datastructure is included in the generated output. Dead-code elimination in the compiler will ensure that the `EmlTreesNode` instances are not included in the final binary if only the inline code is used.

```
#include "rfc.h"
#include <stdio.h>

int main() {
    const int n_features = 8;
    int32_t v[n_features];

    const int pred = rfc_predict(v, n_features);
    printf("pred: %d\n", pred);

    return 0;
}
```

```
gcc -std=c99 -o rfc test.c -I $(python -c "import emlearn; print(emlearn.includedir)") && ./rfc 
```
Out
```
pred: 0
```

skjerns, 2019-05-16:
That works! Thanks!

Would be great if these explanations and also how to compile the code would be written down somewhere. I didn't know about the `emlearn.inlcudedir`

closed 2019-05-16 by skjerns