emlearn/emlearn #9: RandomForestClassifier only works with int

skjerns, 2019-05-16:
Thank's to #8 I can finally compile the code :)

Now I see that the exporting seems to work only with features trained on `int`, is that correct?

Is there any way to easily adapt this to `float/double`?

```
import numpy as np
import sklearn_porter
import matplotlib.pyplot as plt
import emlearn
from sklearn.ensemble import RandomForestClassifier

train_x = np.random.rand(1000, 8)
train_y = np.random.randint(0, 4, 1000)
clf = RandomForestClassifier(n_estimators=51)
clf.fit(train_x, train_y)
    
cmodel = emlearn.convert(clf)  
cmodel.predict(train_x) # predicts only nonsense because train_x is double
```

jonnor, 2019-05-20:
Yeah float should be the primary data format, with uint32/16/8 as the secondary.

I expect to get around to that next week

skjerns, 2019-05-21:
great! if I can be of any help let me know.

jonnor, 2019-06-07:
@skjerns can you test if this works for you now in master/0.9.0?

closed 2019-06-07 by jonnor

skjerns, 2019-06-07:
Thanks very much :)

It seems to be majorly working!

However, I seem to have the same problem as in https://github.com/nok/sklearn-porter/issues/52 .
(I assume it's the same issue as I have the exact same label predictions wrong as when using sklearn-porter. Using `m2cgen` it works fine.)

I think there is some inconsistency when calculating the final probability with a tree that is not fully grown (`max_depth!=None`). The simple calculation of winning classes per tree works in most cases, but the actual computations depends on the depth of the trees and therefore deviates in a few border cases, where the actual probabilities per tree need to be calculated and averaged.

PR #11 (2019-06-07): trees: Default to using float datatype
Start of making it configurable through convert()

Fixes #9 
Closes #10