emlearn/emlearn #12: RFC: Inconsistent results when using `max_depth`

skjerns, 2019-06-28:
As mentioned in #9 , when limiting the depth of a RFC using `max_depth`, there are inconsistencies in the predictions. I've discussed the same issue here: https://github.com/nok/sklearn-porter/issues/52 

I've dived a bit into the issue and figured that it is indeed what I expected: The final leave probabilities are only 0 and 1 if the tree if fully grown.

https://stackoverflow.com/questions/30814231/using-the-predict-proba-function-of-randomforestclassifier-in-the-safe-and-rig
 (see comments)

> 1) About prediction precision: I insist but this is not a question of number of trees. Even with a single decision tree you should be able to get probability predicitions with more than one digits. A decision tree aims at clustering the inputs based on some rules (the decision), and these clusters are the leafs of the tree. If you have a leaf with 2 non-spam emails and one spam email from your training data, then the probability prediction for any email that belongs to this leaf/cluster (with regards to the rules established by fitting the model), is : 1/3 for spam and 2/3 for non-spam. – Sebastien Jun 20 '15 at 14:49
>2) About the dependencies in predictions: Again Sklearn definition gives the answer : the probability is computed with regards to the leaf (corresponding to your email to test) 's characteristics : the number of instances of each class in this leaf. This is set when your model is fitted, so it only depends on the training data. In conclusion : the result is the probability of instance 1 to spam with 60% whatever the other 9 instances' probabilities are. – Sebastien Jun 20 '15 at 15:00

and 

>So depending on implementation: predicted probability is either (a) the mean terminal leaf probability across all trees or (b) the fraction of trees voting either class. If out-of-bag(OOB) prediction, then only in trees where sample is OOB. For a single fully grown tree, I would guess the predicted probability only could be 0 or 1 for any class, because all terminal nodes are pure(same label). If the single tree is not fully grown and/or more trees are grown, then the predicted probability can be a positive rational number from 0 to 1.

https://stats.stackexchange.com/questions/193424/is-decision-tree-output-a-prediction-or-class-probabilities

As I assume that it is quite a bit of re-work to integrate this changes without losing too much performance (e.g. returning pointers to arrays instead of class indexes by the individual trees), I don't want to force you to fix it just for me or so ;-) I just wanted to mention it here. So it's fine with me if it remains like it is for now.

jonnor, 2019-07-12:
Thanks for reporting this, and with the excellent level of detail. Compatibility is important so we should adresses this somehow. Hope to have a look at it the next couple of weeks

jonnor, 2019-07-12:
@skjerns if you by chance are at Europython2019, then I will host a sprint on emlearn and Machine Learning for MicroPython tomorrow and Sunday

jonnor, 2019-07-25:
Looked into this a bit. The RandomForestClassifier in scikit-learn (at least in 0.20) uses the mean values of the `predict_proba()` results from all trees in order to compute the overall probabilities, and then finally to determine the classification. Probabilistic/soft voting instead of majority voting. There does not seem to be any option to change this.

What happens when there is no depth limit is that the proba array from a tree is always sparse: non-zero only for a single class. But it seems that this should be different to the majority voting case at least for some inputs/models?

So it seems we'd need to implement proba and proba voting if we want to be compatible. Which is annoying, since it takes more space for the model, more RAM and more compute. Values of each proba node is likely unique, it will probably invalidate our leaf pruning, blowing up model size.

Alternative would be to document how to get the same prediction with sklearn, by doing the majority prediction. This would allow people to estimate the accuracy loss by the cheaper voting mechanism.

jonnor, 2023-01-01:
Have given this some thought recently. The leaf data should be the proportion of samples belonging to a particular class. Which means storing a number 0.0-1.0 for each of the classes.

It may be possible to store this in the existing node structure, as there is 32/64/71 bits free there for leaf nodes if one uses the space used for threshold/children/feature data for the split nodes. If so, this can implemented be done with little extra costs.
This is assuming that one allocates N bits for each class, and that one sets a limit of M classes. 8 bits and 8 classes will probably cover a wide range of usecases. One challenge is when there is severe class imbalance, then it is inefficient to allocate the same amount to each class.

To regain the possibility of deduplicating leaves, one could support a configurable quantization on the class proportions.

Setting to 0 bits per class should basically become the same as majority voting. Each leaf only being 1 for one of the classes.

jonnor, 2023-03-16:
Storing the leaf data inside a regular node would require using a union, which is a bit ugly. We could instead store a separate structure for leaf nodes. And for example, take negative values for the left/right child to mean an offset into the leaf nodes instead of decision nodes. The child values are already signed integers (for a reason I cannot recall), so this does not cost any more space than already allocated. Would then not store leafs in the decision node table at all. Besides avoiding the union, this approach allows more freedom in selecting the size of leaf nodes, making it entirely up to the user/model parameters.
For small number of classes (most common), this will even be a reduction in model size.

Such a leaf node table could also be used for the inline code generation code. Cause while the decision nodes should probably stay if statements, we want to return an offset into a leaf table to indicate class proportions. At least when the number of bits exceeds what can be returned by value/register, like 32 bits on typical embedded platforms.

jonnor, 2024-06-09:
Separate leaf storage preparation was merged a while ago (see previous MR).

I have now done some research on this topic. Happening in this repository for now: https://github.com/jonnor/embeddedml/tree/master/handson/openml-18
For the precision needed for class probability values. Seems like 8 bits is sufficient, which is good in terms of staying compact.

For maintaining functioning leaf-deduplication. I have tested using K-means to cluster into a small set of values. This looks to work very well, it seems that using just N=2-16 leaves per class gives good performance in the vast majority of cases. That allows to tweak a size/performance tradeoff on a per-model basis.

Still some verification that needs to be done before implementing, but at least making some good progress :)

PR #117 (2025-07-29): trees: Support soft voting using class proportions
Add (optional) support for soft voting. Fixes #12 - finally :)

Sometimes the hard majority gives unacceptable loss in performance. For those cases, it is now possible to switch to soft voting instead. This also gives smoother probabilities. Of course this comes at the cost of model size, as we must store the class proportion in the leaves, not just a class index. Inference time also goes up a bit, but I expect that impact to be less noticable in most cases.

Enabled by passing leaf_bits=N to convert(), where N is a value between 1-8. This is  [... cut]
closed 2025-07-29 by jonnor