emlearn/emlearn #15: multiple definition of 'enum _EmlNetActivationFunction'

RickyKaku, 2019-08-09:
I trained a 3 input NN model named NN4 and tried to upload to MCU. I wrote in MCU:
#include "NN4.h"
double X1[] ={(100,200,300};
     int DV=0;
     DV=NN4_predict(X1, 3);

The error shows:
N-0808:132:26: error: 'NN4_predict' was not declared in this scope

      DV=NN4_predict(X1, 3);
                          ^
exit status 1
multiple definition of 'enum _EmlNetActivationFunction'

RickyKaku, 2019-08-09:
Also, other errors happened in this file, I tried to make a Random Forest model but still failed. Could you write a more detailed example for the Step3 in your documentation? I am new to arduino. Thank you.

jonnor, 2019-08-16:
Hi @RickyKaku thank you for reporting. This sounds like a bug. I will try to have a look at it next week.

jonnor, 2022-01-25:
There is now a complete minimal example at https://emlearn.readthedocs.io/en/latest/auto_examples/xor.html

closed 2022-01-25 by jonnor