emlearn/emlearn #16: Support Keras.Sequential Conv1D layers?

lieuzhenghong, 2019-08-15:
Any plans to support Keras Conv1D layers in a Sequential model?

[Keras Convolutional layer documentation](https://keras.io/layers/convolutional/)

jonnor, 2019-08-16:
Yes, I would like to support basic 1D, 2D convolution - as well as depthwise-separable convolution. There is an old `cnn` branch with a tiny bit of start on this. However I do not have a lot of time to work on this next couple of months. Contributions welcome!

lieuzhenghong, 2019-08-16:
what's the best way for a contributor to get to grips with the codebase and implement functionality?

jonnor, 2019-08-16:
To implement new functionality I recommend starting by adding a new testcase. For high-level testing of Keras compatibility that would mean adding a new entry to KERAS_MODELS in tests/test_net.py - probably using a new helper function for building a CNN model.

If we look at the codebase from the tests inwards, then the codebase looks something like:

- Automated tests. `test/*.py`
- Model conversion. `emlearn/*.py`
- Python bindings (primarily for testing). `bindings/*.cpp`
- Inference engine `emlearn/eml_*.c`

jonnor, 2019-08-16:
I have not seen Frugally Deep no, thanks for linking it. If C++ is not an issue (some embedded shops are very conservative or have outdated compilers) then it looks like a great option.
I see it has runtime model building from JSON, but presumably one could avoid that by using the low-level APIs.

jonnor, 2019-11-20:
Since [TensorFlow for microcontrollers](https://www.tensorflow.org/lite/microcontrollers) now exists and seems to be in OK shape, I'm seriously considering to say that deep learning models like CNNs etc is out-of-scope for `emlearn`.

jonnor, 2020-03-12:
Here is another solution for CNNs, which follows the goals of `emlearn` very closely. https://github.com/ekut-es/pico-cnn

So I think we don't need to add this in emlearn. We can focus on the classic methods.

closed 2020-03-12 by jonnor

jonnor, 2020-10-28:
[nnom](https://github.com/majianjia/nnom/) is another good implementation of neural networks (including CNNs, RNNs) for microcontrollers

jonnor, 2024-04-26:
https://github.com/sipeed/TinyMaix/ is another good alternative for small CNN models.
It supports up to MobileNets, very simple and compact code.