emlearn/emlearn #17: convert() fails under windows

micooke, 2019-11-14:
the convert() function creates a CompiledClassifier() class object, that calls build_classifier() in its initialisation. It does not pass a compiler parameter, so unless the build machine has gcc installed it will always fail at this point?

Im going to fix this in my forked repo by back-propogating 'compiler' up to convert(), but if you have a better alternative i would be happy to create the PR to fix it.

Cheers,Mark

jonnor, 2019-11-14:
Confirming that this is an issue that should be fixed somehow.

This kind of stuff should have defaults that work on most platforms (Win, Linux, OSX). With configuration from outside possible to handle any edge-cases.

jonnor, 2019-11-14:
Semi-related, I would love to have Windows (and Mac OSX) enabled in Continuous Integration, to ensure that things work on those platforms as well.

jonnor, 2019-11-23:
@micooke Windows fixes have been released to PyPi in emlearn 0.10.1. Thanks a lot for your contributions! 

If possible, please test whether the release works as expected on Windows. As we don't build and publish binary wheels, a working C compiler is still needed though.

micooke, 2019-11-25:
It works!
Buut you are going to hate me :/ It builds under msvc, but in order for it to compile under e.g. cygwin, mingw32,bcpp we would need to re-propogate the compiler flag, or maybe add a global to common.py with a ```set_compiler()``` function?
Then we would have the issue of checking this option under pytest?
Is this something you want to support?

micooke, 2019-11-25:
either way, ill close this as it works under windows!

closed 2019-11-25 by micooke

jonnor, 2019-11-25:
Thanks for fixing this!

micooke, 2019-11-25:
Thanks for being patient with me :smile: