emlearn/emlearn #23: Add Windows to CI

jonnor, 2019-11-21:
MSCV on Windows has some minor but notable differences to GCC (which is tested on my computer and on Travis CI). So would be great to protect against the build breaking on Windows by adding CI for that.

Appveyour and Circle CI are options here, I think

micooke, 2019-11-21:
Travis CI also supports linux, macos and windoze https://blog.travis-ci.com/2018-10-11-windows-early-release
I havent tested this, but it looks to be fairly easy to use. Automating the msvc, and python installs might be fun under win though 🤯🔫

jonnor, 2019-11-21:
Ah great they have Window now. Actually setting up Python on Windows is super easy if one uses Conda (miniconda) - but I dont actually know which C compiler that uses / brings in, or if that it is up to the end-user.

micooke, 2019-11-21:
Anaconda with MSVC 2019 was my build env. Distutils required that version for Python 3.5+ https://wiki.python.org/moin/WindowsCompilers

micooke, 2019-11-21:
Oh neat, I just read this https://docs.travis-ci.com/user/reference/windows/
The windows env has bash and vs2017

PR #25 (2019-11-23): Add Windows to Travis CI
Fixes #23
closed 2019-11-23 by jonnor