emlearn/emlearn #26: Error when installing in Windows

ramonjsi, 2020-04-20:
C:\WPy64-3760\scripts>pip install --user emlearn
Collecting emlearn
  Using cached https://files.pythonhosted.org/packages/01/18/8f05dbf9240d0d3cee5470456fa5c370858914b9f7885373c506d0cff398/emlearn-0.10.1.tar.gz
Requirement already satisfied: pybind11>=2.2 in c:\wpy64-3760\python-3.7.6.amd64\lib\site-packages (from emlearn) (2.4.3)
Requirement already satisfied: numpy>=1.14.0 in c:\wpy64-3760\python-3.7.6.amd64\lib\site-packages (from emlearn) (1.17.4+mkl)
Building wheels for collected packages: emlearn
  Building wheel for emlearn (setup.py) ... error
  ERROR: Command errored out with exit status 1:
   command: 'C:\WPy64-3760\python-3.7.6.amd64\python.exe' -u -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'C:\\Users\\admin\\AppData\\Local\\Temp\\pip-install-6eupeo_3\\emlearn\\setup.py'"'"'; __file__='"'"'C:\\Users\\admin\\AppData\\Local\\Temp\\pip-install-6eupeo_3\\emlearn\\setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' bdist_wheel -d 'C:\Users\admin\AppData\Local\Temp\pip-wheel-hidbzz02' --python-tag cp37
       cwd: C:\Users\admin\AppData\Local\Temp\pip-install-6eupeo_3\emlearn\
  Complete output (42 lines):
  running bdist_wheel
  running build
  running build_py
  creating build
  creating build\lib.win-amd64-3.7
  creating build\lib.win-amd64-3.7\emlearn
  copying emlearn\bayes.py -> build\lib.win-amd64-3.7\emlearn
  copying emlearn\cgen.py -> build\lib.win-amd64-3.7\emlearn
  copying emlearn\common.py -> build\lib.win-amd64-3.7\emlearn
  copying emlearn\convert.py -> build\lib.win-amd64-3.7\emlearn
  copying emlearn\net.py -> build\lib.win-amd64-3.7\emlearn
  copying emlearn\signal.py -> build\lib.win-amd64-3.7\emlearn
  copying emlearn\trees.py -> build\lib.win-amd64-3.7\emlearn
  copying emlearn\__init__.py -> build\lib.win-amd64-3.7\emlearn
  running egg_info
  writing emlearn.egg-info\PKG-INFO
  writing dependency_links to emlearn.egg-info\dependency_links.txt
  writing requirements to emlearn.egg-info\requires.txt
  writing top-level names to emlearn.egg-info\top_level.txt
  reading manifest file 'emlearn.egg-info\SOURCES.txt'
  reading manifest template 'MANIFEST.in'
  writing manifest file 'emlearn.egg-info\SOURCES.txt'
  copying emlearn\eml_audio.h -> build\lib.win-amd64-3.7\emlearn
  copying emlearn\eml_bayes.h -> build\lib.win-amd64-3.7\emlearn
  copying emlearn\eml_benchmark.h -> build\lib.win-amd64-3.7\emlearn
  copying emlearn\eml_common.h -> build\lib.win-amd64-3.7\emlearn
  copying emlearn\eml_fft.h -> build\lib.win-amd64-3.7\emlearn
  copying emlearn\eml_fixedpoint.h -> build\lib.win-amd64-3.7\emlearn
  copying emlearn\eml_iir.h -> build\lib.win-amd64-3.7\emlearn
  copying emlearn\eml_net.h -> build\lib.win-amd64-3.7\emlearn
  copying emlearn\eml_test.h -> build\lib.win-amd64-3.7\emlearn
  copying emlearn\eml_trees.h -> build\lib.win-amd64-3.7\emlearn
  copying emlearn\eml_vector.h -> build\lib.win-amd64-3.7\emlearn
  running build_ext
  building 'eml_trees' extension
  creating build\temp.win-amd64-3.7
  creating build\temp.win-amd64-3.7\Release
  creating build\temp.win-amd64-3.7\Release\bindings
  C:\Program Files (x86)\Microsoft Visual Studio\2019\BuildTools\VC\Tools\MSVC\14.25.28610\bin\HostX86\x64\cl.exe /c /nologo /Ox /W3 /GL /DNDEBUG /MD -Iemlearn/ -IC:\WPy64-3760\python-3.7.6.amd64\Include -IC:\Users\admin\AppData\Roaming\Python\Python37\Include -IC:\WPy64-3760\python-3.7.6.amd64\include -IC:\WPy64-3760\python-3.7.6.amd64\include "-IC:\Program Files (x86)\Microsoft Visual Studio\2019\BuildTools\VC\Tools\MSVC\14.25.28610\include" /EHsc /Tpbindings/eml_trees.cpp /Fobuild\temp.win-amd64-3.7\Release\bindings/eml_trees.obj /EHsc /DVERSION_INFO=\"0.10.1\"
  eml_trees.cpp
  bindings/eml_trees.cpp(2): fatal error C1083: No se puede abrir el archivo incluir: 'stdio.h': No such file or directory
  error: command 'C:\\Program Files (x86)\\Microsoft Visual Studio\\2019\\BuildTools\\VC\\Tools\\MSVC\\14.25.28610\\bin\\HostX86\\x64\\cl.exe' failed with exit status 2
  ----------------------------------------
  ERROR: Failed building wheel for emlearn
  Running setup.py clean for emlearn
Failed to build emlearn

ramonjsi, 2020-04-20:
Hi,

I have installed Visual Studio Build tools 2019 and I get this error, any clue?

Thank in advance.

ramonjsi, 2020-04-21:
Solved by installing the packages in the figure.

[image]

closed 2020-04-21 by ramonjsi

jonnor, 2020-06-01:
Glad that you found a solution. And thanks for documenting it for others to benefit. Muchas gracias!