emlearn/emlearn #31: RandomForestRegressor

jessie831024, 2021-02-10:
Can you add feature support for RandomForestRegressor?

jonnor, 2021-02-10:
Hi! Thank for the request. A couple of others have asked for the same. Also related is #14.

In the coming weeks I do not think that I will have time to implement this, unfortunately. Contributions are very welcomed!

jonnor, 2021-12-31:
Stumbled over this very simple implementation, https://data-and-the-world.onrender.com/posts/python-random-forest-c/

kkmoslehpour, 2022-12-08:
Also looking for support for RandomForestRegressor

jonnor, 2022-12-15:
Implemented now in https://github.com/emlearn/emlearn/pull/54

jonnor, 2023-04-16:
This feature was released in emlearn 0.16.3 on 01.01.2023: https://github.com/emlearn/emlearn/releases/tag/0.16.3

closed 2023-04-16 by jonnor