emlearn/emlearn #33: Concerning to generated code size

NatakuG, 2021-05-18:
Hi,
I am using emlearn to generate Decision Tree and Random Forest modules for an embedded system on Zephyr. 
For each DT, it generates around 25 kB for .h file and 15 kB for .o file. For RF of 10 estimators, it generates about 240 kB for .h and about 100 kB for .o.
Is there any other method that I can do to optimize from the code size ? For example, I noticed that the generated code uses floating-point or double-precision format, which may consume more memory for an embedded system.

jonnor, 2021-05-28:
Hi @NatakuG 

Are you using eml_trees_predict(...) or the generated function sometree_predict() ? Which compiler and optimization flags do you use? It is critical to have something like -O3 optimization with dead-code elimination enable.

The default code for trees uses floating point. Doubles should never be used, that would be a bug, so please report if you see that anywhere.

There exists fixed-point support for trees, which can be enabled using `emlearn.convert(..., dtype='uint8_t')`

Limiting the depth of trees is important to keep them reasonably sized. I recommend using the scikit-learn parameters `min_samples_leaf` or `min_samples_split` for this. `max_depth` is another alternative, but it has very large step size and does not allow for deeper trees where needed.

NatakuG, 2021-06-07:
Thank you so much for your reply. I will try it out 👍🏼

jonnor, 2021-12-27:
How did it go?

jonnor, 2023-01-01:
There is now an example of hyperparameter tuning to make small tree ensembles, at https://emlearn.readthedocs.io/en/latest/auto_examples/trees_hyperparameters.html

jonnor, 2023-06-10:
There are now some more documentation at https://emlearn.readthedocs.io/en/latest/tree_based_models.html

Closing this one. If there are specific questions, please open a new issue.

closed 2023-06-10 by jonnor