emlearn/emlearn #34: Support IsolationForest

jonnor, 2021-05-28:
IsolationForest is a simple but very effective outlier/anomaly detection method.

This would require being able to compute the decision paths, or at least the depth of the trees.

scikit-learn implementation reference, https://github.com/scikit-learn/scikit-learn/blob/3a64fecd1f1d30a17998b254f94613adee48a930/sklearn/ensemble/_iforest.py#L423

jonnor, 2022-09-20:
The first step of this would be to support inference. But training of an IsolationForest should also be doable for many devices, and can be very useful to implement on-edge anomaly detection - which can be very useful to minimize data transfers in IoT sensor systems. So that could maybe be in scope also for emlearn.

jonnor, 2023-01-01:
Wrote some notes at https://github.com/jonnor/embeddedml/blob/master/emlearn/isolation-forest.md

jonnor, 2024-04-20:
A TinyML focused papers on improved IsolationForest variant:
https://www.sciencedirect.com/science/article/abs/pii/S0020025522008155
https://github.com/tombarba/TinyWeaklyIsolationForest

smokedlab, 2026-11-06:
Hi @jonnor,

I'd be interested in working on IsolationForest support as a follow-up to my GradientBoostingClassifier contribution (#123).

I'm particularly interested in embedded anomaly detection for industrial condition monitoring.

I'd like to start with scikit-learn IsolationForest inference, including path length calculation, anomaly scores, and tests against scikit-learn. I'll also review the implementation notes and references already shared in this issue.

Would this scope make sense for an initial PR? I'd be happy to work on it.

Cheers, Hadi

jonnor, 2026-11-10:
Hi @smokedlab. Yes an implementation that supports converting scikit-learn IsolationForest and implements inference would be a good start!

smokedlab, 2026-11-11:
Thanks  @jonnor !

I'll start by looking into the existing tree implementation and work on adding IsolationForest conversion and inference support.

Cheers,