emlearn/emlearn #35: Issue installing emlearn on macos

Anubhav2017, 2021-07-21:
I am trying to install emlearn in a conda environment as suggested in the README using pip install  emlearn. However I am getting the folowing error. I am using Macbook Air M1 running MacOS Big-Sur 11.4.

anubhav@Anubhavs-MacBook-Air emlearn % pip install emlearn       
Collecting emlearn
  Using cached emlearn-0.12.2.tar.gz (41 kB)
Requirement already satisfied: pybind11>=2.2 in /Users/anubhav/.local/lib/python3.9/site-packages (from emlearn) (2.7.0)
Requirement already satisfied: numpy>=1.14.0 in /Users/anubhav/.local/lib/python3.9/site-packages (from emlearn) (1.21.1)
Building wheels for collected packages: emlearn
  Building wheel for emlearn (setup.py) ... error
  ERROR: Command errored out with exit status 1:
   command: /Users/anubhav/miniforge3/envs/ddpenv/bin/python3.9 -u -c 'import io, os, sys, setuptools, tokenize; sys.argv[0] = '"'"'/private/var/folders/st/g0z4v7q50sj7prs3cygcjp6m0000gn/T/pip-install-mnpv4g4_/emlearn_7a9699e8d6814e579ea70db1b4600682/setup.py'"'"'; __file__='"'"'/private/var/folders/st/g0z4v7q50sj7prs3cygcjp6m0000gn/T/pip-install-mnpv4g4_/emlearn_7a9699e8d6814e579ea70db1b4600682/setup.py'"'"';f = getattr(tokenize, '"'"'open'"'"', open)(__file__) if os.path.exists(__file__) else io.StringIO('"'"'from setuptools import setup; setup()'"'"');code = f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' bdist_wheel -d /private/var/folders/st/g0z4v7q50sj7prs3cygcjp6m0000gn/T/pip-wheel-wyqc3i63
       cwd: /private/var/folders/st/g0z4v7q50sj7prs3cygcjp6m0000gn/T/pip-install-mnpv4g4_/emlearn_7a9699e8d6814e579ea70db1b4600682/
  Complete output (238 lines):
  running bdist_wheel
  running build
  running build_py
  creating build
  creating build/lib.macosx-11.0-arm64-3.9
  creating build/lib.macosx-11.0-arm64-3.9/emlearn
  copying emlearn/version.py -> build/lib.macosx-11.0-arm64-3.9/emlearn
  copying emlearn/signal.py -> build/lib.macosx-11.0-arm64-3.9/emlearn
  copying emlearn/convert.py -> build/lib.macosx-11.0-arm64-3.9/emlearn
  copying emlearn/net.py -> build/lib.macosx-11.0-arm64-3.9/emlearn
  copying emlearn/__init__.py -> build/lib.macosx-11.0-arm64-3.9/emlearn
  copying emlearn/cgen.py -> build/lib.macosx-11.0-arm64-3.9/emlearn
  copying emlearn/common.py -> build/lib.macosx-11.0-arm64-3.9/emlearn
  copying emlearn/bayes.py -> build/lib.macosx-11.0-arm64-3.9/emlearn
  copying emlearn/trees.py -> build/lib.macosx-11.0-arm64-3.9/emlearn
  creating build/lib.macosx-11.0-arm64-3.9/emlearn/tools
  copying emlearn/tools/mel_filterbank.py -> build/lib.macosx-11.0-arm64-3.9/emlearn/tools
  copying emlearn/tools/window_function.py -> build/lib.macosx-11.0-arm64-3.9/emlearn/tools
  copying emlearn/tools/__init__.py -> build/lib.macosx-11.0-arm64-3.9/emlearn/tools
  running egg_info
  writing emlearn.egg-info/PKG-INFO
  writing dependency_links to emlearn.egg-info/dependency_links.txt
  writing entry points to emlearn.egg-info/entry_points.txt
  writing requirements to emlearn.egg-info/requires.txt
  writing top-level names to emlearn.egg-info/top_level.txt
  reading manifest file 'emlearn.egg-info/SOURCES.txt'
  reading manifest template 'MANIFEST.in'
  writing manifest file 'emlearn.egg-info/SOURCES.txt'
  copying emlearn/VERSION -> build/lib.macosx-11.0-arm64-3.9/emlearn
  copying emlearn/eml_audio.h -> build/lib.macosx-11.0-arm64-3.9/emlearn
  copying emlearn/eml_bayes.h -> build/lib.macosx-11.0-arm64-3.9/emlearn
  copying emlearn/eml_benchmark.h -> build/lib.macosx-11.0-arm64-3.9/emlearn
  copying emlearn/eml_common.h -> build/lib.macosx-11.0-arm64-3.9/emlearn
  copying emlearn/eml_fft.h -> build/lib.macosx-11.0-arm64-3.9/emlearn
  copying emlearn/eml_fixedpoint.h -> build/lib.macosx-11.0-arm64-3.9/emlearn
  copying emlearn/eml_iir.h -> build/lib.macosx-11.0-arm64-3.9/emlearn
  copying emlearn/eml_net.h -> build/lib.macosx-11.0-arm64-3.9/emlearn
  copying emlearn/eml_test.h -> build/lib.macosx-11.0-arm64-3.9/emlearn
  copying emlearn/eml_trees.h -> build/lib.macosx-11.0-arm64-3.9/emlearn
  copying emlearn/eml_vector.h -> build/lib.macosx-11.0-arm64-3.9/emlearn
  running build_ext
  creating var
  creating var/folders
  creating var/folders/st
  creating var/folders/st/g0z4v7q50sj7prs3cygcjp6m0000gn
  creating var/folders/st/g0z4v7q50sj7prs3cygcjp6m0000gn/T
  clang -Wno-unused-result -Wsign-compare -Wunreachable-code -DNDEBUG -fwrapv -O2 -Wall -fPIC -O2 -isystem /Users/anubhav/miniforge3/envs/ddpenv/include -arch arm64 -fPIC -O2 -isystem /Users/anubhav/miniforge3/envs/ddpenv/include -arch arm64 -I/Users/anubhav/miniforge3/envs/ddpenv/include/python3.9 -c /var/folders/st/g0z4v7q50sj7prs3cygcjp6m0000gn/T/tmpiio6epr1.cpp -o var/folders/st/g0z4v7q50sj7prs3cygcjp6m0000gn/T/tmpiio6epr1.o -std=c++14
  clang -Wno-unused-result -Wsign-compare -Wunreachable-code -DNDEBUG -fwrapv -O2 -Wall -fPIC -O2 -isystem /Users/anubhav/miniforge3/envs/ddpenv/include -arch arm64 -fPIC -O2 -isystem /Users/anubhav/miniforge3/envs/ddpenv/include -arch arm64 -I/Users/anubhav/miniforge3/envs/ddpenv/include/python3.9 -c /var/folders/st/g0z4v7q50sj7prs3cygcjp6m0000gn/T/tmppz09p62t.cpp -o var/folders/st/g0z4v7q50sj7prs3cygcjp6m0000gn/T/tmppz09p62t.o -fvisibility=hidden
  building 'eml_trees' extension
  creating build/temp.macosx-11.0-arm64-3.9
  creating build/temp.macosx-11.0-arm64-3.9/bindings
  clang -Wno-unused-result -Wsign-compare -Wunreachable-code -DNDEBUG -fwrapv -O2 -Wall -fPIC -O2 -isystem /Users/anubhav/miniforge3/envs/ddpenv/include -arch arm64 -fPIC -O2 -isystem /Users/anubhav/miniforge3/envs/ddpenv/include -arch arm64 -Iemlearn/ -I/Users/anubhav/.local/lib/python3.9/site-packages/pybind11/include -I/Users/anubhav/.local/lib/python3.9/site-packages/pybind11/include -I/Users/anubhav/miniforge3/envs/ddpenv/include/python3.9 -c bindings/eml_trees.cpp -o build/temp.macosx-11.0-arm64-3.9/bindings/eml_trees.o -stdlib=libc++ -mmacosx-version-min=10.7 -DVERSION_INFO="0.12.2" -std=c++14 -fvisibility=hidden
  In file included from bindings/eml_trees.cpp:6:
  In file included from /Users/anubhav/.local/lib/python3.9/site-packages/pybind11/include/pybind11/pybind11.h:44:
  In file included from /Users/anubhav/.local/lib/python3.9/site-packages/pybind11/include/pybind11/attr.h:13:
  In file included from /Users/anubhav/.local/lib/python3.9/site-packages/pybind11/include/pybind11/cast.h:13:
  In file included from /Users/anubhav/.local/lib/python3.9/site-packages/pybind11/include/pybind11/pytypes.h:12:
[... 401 lines cut ...]
    In file included from bindings/eml_trees.cpp:6:
    In file included from /Users/anubhav/.local/lib/python3.9/site-packages/pybind11/include/pybind11/pybind11.h:44:
    In file included from /Users/anubhav/.local/lib/python3.9/site-packages/pybind11/include/pybind11/attr.h:13:
    In file included from /Users/anubhav/.local/lib/python3.9/site-packages/pybind11/include/pybind11/cast.h:13:
    In file included from /Users/anubhav/.local/lib/python3.9/site-packages/pybind11/include/pybind11/pytypes.h:12:
    In file included from /Users/anubhav/.local/lib/python3.9/site-packages/pybind11/include/pybind11/detail/common.h:159:
    In file included from /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX.sdk/usr/include/c++/v1/forward_list:184:
    In file included from /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX.sdk/usr/include/c++/v1/memory:681:
    /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX.sdk/usr/include/c++/v1/atomic:1052:73: error: unknown type name 'ptrdiff_t'
    _Tp* __cxx_atomic_fetch_add(__cxx_atomic_base_impl<_Tp*> volatile* __a, ptrdiff_t __delta, memory_order __order) _NOEXCEPT {
                                                                            ^
    /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX.sdk/usr/include/c++/v1/atomic:1057:65: error: unknown type name 'ptrdiff_t'
    _Tp* __cxx_atomic_fetch_add(__cxx_atomic_base_impl<_Tp*> * __a, ptrdiff_t __delta, memory_order __order) _NOEXCEPT {
                                                                    ^
    /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX.sdk/usr/include/c++/v1/atomic:1073:73: error: unknown type name 'ptrdiff_t'
    _Tp* __cxx_atomic_fetch_sub(__cxx_atomic_base_impl<_Tp*> volatile* __a, ptrdiff_t __delta, memory_order __order) _NOEXCEPT {
[... cut ...] pen)(__file__) if os.path.exists(__file__) else io.StringIO('"'"'from setuptools import setup; setup()'"'"');code = f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' install --record /private/var/folders/st/g0z4v7q50sj7prs3cygcjp6m0000gn/T/pip-record-mzwax3f0/install-record.txt --single-version-externally-managed --compile --install-headers /Users/anubhav/miniforge3/envs/ddpenv/include/python3.9/emlearn Check the logs for full command output.

jonnor, 2021-07-21:
Hi @Anubhav2017 and thank you for the issue report.

It seems the ptrdiff_t type, which is used by pybind11 is not found in the system headers. It is usually in the stddef.h header, but it seems like on Mac OS this is not always the case, ref https://github.com/nigels-com/glew/issues/219#issuecomment-473609147
What suprises me is that no-one else has reported this issue with pybind11 on Mac OS, as there does not seem to be anything emlearn specific here.

jonnor, 2021-07-21:
Can you test with Python 3.8 instead of 3.9? Do you have access to a non-M1 Mac to test this on?

bpaser, 2021-09-23:
I'm having the same issue using Python 3.8 on windows 10. I switched to Linux Ubuntu and it was able to install.

Brax94, 2021-11-15:
Encountered the same issue with similar error message. 

Python Version: 3.8
Pip version:  20.2.4
Mac OSX Version: Big Sur 11.4
GCC version from output: 
    gcc -v  
    Configured with: --prefix=/Library/Developer/CommandLineTools/usr --with-gxx-include-dir=/Library/Developer/CommandLineTools/SDKs/MacOSX.sdk/usr/include/c++/4.2.1
    Apple clang version 12.0.5 (clang-1205.0.22.9)
    Target: x86_64-apple-darwin20.5.0
    Thread model: posix
    InstalledDir: /Library/Developer/CommandLineTools/usr/bin

ARM gcc version:
  arm-none-eabi-gcc --version
  arm-none-eabi-gcc (GNU Arm Embedded Toolchain 10.3-2021.10) 10.3.1 20210824 (release)

jonnor, 2021-11-15:
What is the version of pybind11?

pip show pybind11 | grep Version
python -c "import pybind11; print(pybind11.__version__)"

Brax94, 2021-11-15:
This might be relevant: 
https://githubmemory.com/repo/nanodbc/nanodbc/issues/274
https://bugs.llvm.org/show_bug.cgi?id=42540

Brax94, 2021-11-15:
pip show pybind11 | grep Version
Version: 2.8.1

jonnor, 2021-11-15:
Good find on that LLVM issue - we indeed have a file called VERSION, which is put next to other .h files, so it is likely on the include path. Pretty good chance that is the source of the problems.
Renaming the file to be VERSION.txt should then fix it. Want to try to make a merge request with such a fix? Note that the VERSION file is referenced a couple of different places

jonnor, 2021-11-16:
Fixed now thanks to @Brax94 . Confirmed on 2 Mac machines

closed 2021-11-16 by jonnor