emlearn/emlearn #39: Support to multi-label problems

armandokeller, 2021-09-09:
It does support multi-label classification?

I'm using RandomForestClassifier to solve a multi-label problem (120 features and 14 labels) and it works fine in python.

When I try to run conversion:
```python
import emlearn

cmodel = emlearn.convert(rfmodel, method="inline")
cmodel.save("randomforest.h")
```
I got this error:
```
AssertionError                            Traceback (most recent call last)
<ipython-input-12-cb201d9f8f8f> in <module>()
      1 import emlearn
      2 
----> 3 cmodel = emlearn.convert(rfmodel,method="inline")
      4 cmodel.save(file='rfecg.h')

3 frames
/usr/local/lib/python3.7/dist-packages/emlearn/convert.py in convert(estimator, kind, method, dtype)
     12     # Use name instead of instance to avoid hard dependency on the libraries
     13     if kind in ['RandomForestClassifier', 'ExtraTreesClassifier', 'DecisionTreeClassifier']:
---> 14         return trees.Wrapper(estimator, method, dtype=dtype)
     15     elif kind == 'MLPClassifier':
     16         return net.convert_sklearn_mlp(estimator, method)

/usr/local/lib/python3.7/dist-packages/emlearn/trees.py in __init__(self, estimator, classifier, dtype)
    323             trees = [ estimator.tree_ ]
    324 
--> 325         self.forest_ = flatten_forest(trees)
    326         self.forest_ = remove_duplicate_leaves(self.forest_)
    327 

/usr/local/lib/python3.7/dist-packages/emlearn/trees.py in flatten_forest(trees)
     37 
     38     for tree in trees:
---> 39         flat = flatten_tree(tree)
     40 
     41         # Offset the nodes in tree, so they can be stored in one array

/usr/local/lib/python3.7/dist-packages/emlearn/trees.py in flatten_tree(tree)
     17 
     18     assert tree.node_count == len(tree.value)
---> 19     assert tree.value.shape[1] == 1 # number of outputs
     20 
     21     for left, right, feature, th, value in zip(tree.children_left, tree.children_right, tree.feature, tree.threshold, tree.value):

AssertionError: 
```

jonnor, 2021-09-09:
Hi @armandokeller - thank you for reporting. Multi-output / multi-output is currently not supported. The work-around would be to use dedicated models for each output. 
Contributions to add support for this is welcomed, as long as it does not hurt the performance of the single-output case (which is much more typical for small systems).

jonnor, 2023-10-29:
The neural network inference in emlearn supports multi-label/multi-output out-of-the-box, so that is an alternative here.

jonnor, 2026-01-11:
There are now helpers in scikit-learn for multi-output, see https://scikit-learn.org/stable/modules/generated/sklearn.multioutput.MultiOutputClassifier.html and https://scikit-learn.org/stable/modules/generated/sklearn.multioutput.MultiOutputRegressor.html

So we will not add support directly in the tree-based models. Examples on how to use the above utilities with emlearn would be appreciated, and maybe also to give a bit nicer / more helpful error message.