emlearn/emlearn #40: Unable to install emlearn on Windows10

bpaser, 2021-09-23:
I'm trying to install emlearn, I'm using python3.8 on windows 10. attatched is the error message that I get. I've tried different versions of python but nothing seems to give.

[error_message.txt](https://github.com/emlearn/emlearn/files/7214505/error_message.txt)

bpaser, 2021-09-23:
I was able to get it to work by downloading version 0.4.1, I did have to replace the code for bayes.py with the newer version to get it to work.

jonnor, 2021-09-25:
Hi and thank you for the report!
Version 0.4.1 is several years old. We should try to fix this issue on recent versions. Reopening the issue.

Here is the error from your log. It may be that most of it is just a follow from 
   emlearn/version(1): error C2059: syntax error: 'constant'

If others have this issue on Windows, please comment.

```
    C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\VC\Tools\MSVC\14.26.28801\bin\HostX86\x64\cl.exe /c /nologo /Ox /W3 /GL /DNDEBUG /MD -Iemlearn/ -Ic:\users\bpase\anaconda3\lib\site-packages\pybind11\include -Ic:\users\bpase\anaconda3\lib\site-packages\pybind11\include -Ic:\users\bpase\anaconda3\include -Ic:\users\bpase\anaconda3\include "-IC:\Program Files (x86)\Microsoft Visual Studio\2019\Community\VC\Tools\MSVC\14.26.28801\ATLMFC\include" "-IC:\Program Files (x86)\Microsoft Visual Studio\2019\Community\VC\Tools\MSVC\14.26.28801\include" "-IC:\Program Files (x86)\Windows Kits\NETFXSDK\4.8\include\um" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.18362.0\ucrt" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.18362.0\shared" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.18362.0\um" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.18362.0\winrt" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.18362.0\cppwinrt" /EHsc /Tpbindings/eml_trees.cpp /Fobuild\temp.win-amd64-3.8\Release\bindings/eml_trees.obj /EHsc /DVERSION_INFO=\"0.12.2\"
    eml_trees.cpp
    emlearn/eml_trees.h(56): warning C4244: 'return': conversion from 'float' to 'int32_t', possible loss of data
    emlearn/version(1): error C2059: syntax error: 'constant'
    c:\users\bpase\anaconda3\lib\site-packages\pybind11\include\pybind11\detail/common.h(378): error C2143: syntax error: missing ';' before '{'
    c:\users\bpase\anaconda3\lib\site-packages\pybind11\include\pybind11\detail/common.h(378): error C2447: '{': missing function header (old-style formal list?)
    c:\users\bpase\anaconda3\lib\site-packages\pybind11\include\pybind11\buffer_info.h(53): error C2039: 'any_container': is not a member of 'pybind11::detail'
    c:\users\bpase\anaconda3\lib\site-packages\pybind11\include\pybind11\buffer_info.h(16): note: see declaration of 'pybind11::detail'
    c:\users\bpase\anaconda3\lib\site-packages\pybind11\include\pybind11\buffer_info.h(52): error C2061: syntax error: identifier 'any_container'
    c:\users\bpase\anaconda3\lib\site-packages\pybind11\include\pybind11\buffer_info.h(63): error C2039: 'any_container': is not a member of 'pybind11::detail'
    c:\users\bpase\anaconda3\lib\site-packages\pybind11\include\pybind11\buffer_info.h(16): note: see declaration of 'pybind11::detail'
    c:\users\bpase\anaconda3\lib\site-packages\pybind11\include\pybind11\buffer_info.h(63): error C2061: syntax error: identifier 'any_container'
    c:\users\bpase\anaconda3\lib\site-packages\pybind11\include\pybind11\buffer_info.h(120): error C2039: 'any_container': is not a member of 'pybind11::detail'
    c:\users\bpase\anaconda3\lib\site-packages\pybind11\include\pybind11\buffer_info.h(16): note: see declaration of 'pybind11::detail'
    c:\users\bpase\anaconda3\lib\site-packages\pybind11\include\pybind11\buffer_info.h(119): error C2061: syntax error: identifier 'any_container'
    c:\users\bpase\anaconda3\lib\site-packages\pybind11\include\pybind11\buffer_info.h(55): error C2065: 'shape_in': undeclared identifier
    c:\users\bpase\anaconda3\lib\site-packages\pybind11\include\pybind11\buffer_info.h(55): error C2065: 'strides_in': undeclared identifier
    c:\users\bpase\anaconda3\lib\site-packages\pybind11\include\pybind11\buffer_info.h(57): error C3861: 'pybind11_fail': identifier not found
    c:\users\bpase\anaconda3\lib\site-packages\pybind11\include\pybind11\buffer_info.h(67): error C2661: 'pybind11::buffer_info::buffer_info': no overloaded function takes 7 arguments
    c:\users\bpase\anaconda3\lib\site-packages\pybind11\include\pybind11\buffer_info.h(78): error C2661: 'pybind11::buffer_info::buffer_info': no overloaded function takes 7 arguments
    c:\users\bpase\anaconda3\lib\site-packages\pybind11\include\pybind11\buffer_info.h(121): error C2065: 'shape_in': undeclared identifier
    c:\users\bpase\anaconda3\lib\site-packages\pybind11\include\pybind11\buffer_info.h(121): error C2065: 'strides_in': undeclared identifier
    c:\users\bpase\anaconda3\lib\site-packages\pybind11\include\pybind11\buffer_info.h(135): error C2039: 'enable_if_t': is not a member of 'pybind11::detail'
    c:\users\bpase\anaconda3\lib\site-packages\pybind11\include\pybind11\buffer_info.h(127): note: see declaration of 'pybind11::detail'
    c:\users\bpase\anaconda3\lib\site-packages\pybind11\include\pybind11\buffer_info.h(135): error C2065: 'enable_if_t': undeclared identifier
    c:\users\bpase\anaconda3\lib\site-packages\pybind11\include\pybind11\buffer_info.h(135): error C2974: 'pybind11::detail::compare_buffer_info': invalid template argument for 'SFINAE', type expected
    c:\users\bpase\anaconda3\lib\site-packages\pybind11\include\pybind11\buffer_info.h(129): note: see declaration of 'pybind11::detail::compare_buffer_info'
    c:\users\bpase\anaconda3\lib\site-packages\pybind11\include\pybind11\buffer_info.h(135): error C2988: unrecognizable template declaration/definition
    c:\users\bpase\anaconda3\lib\site-packages\pybind11\include\pybind11\buffer_info.h(135): error C2143: syntax error: missing ';' before '>'
[... cut ...]
    C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\VC\Tools\MSVC\14.26.28801\include\iterator(24): error C2631: 'front_insert_iterator': a class or enum cannot be defined in an alias template
    bindings/eml_trees.cpp: error C2988: unrecognizable template declaration/definition
    bindings/eml_trees.cpp: error C2059: syntax error: 'namespace'
    bindings/eml_trees.cpp: fatal error C1903: unable to recover from previous error(s); stopping compilation
    error: command 'C:\\Program Files (x86)\\Microsoft Visual Studio\\2019\\Community\\VC\\Tools\\MSVC\\14.26.28801\\bin\\HostX86\\x64\\cl.exe' failed with exit status 2
```

mohneesh7, 2021-10-05:
I am having the same issue

mohneesh7, 2021-10-05:
I installed the 0.11.0 version, That works fine

jonnor, 2021-11-15:
This might actually have the same root cause as #35 (the "version" file being picked up as a C++ header).

jonnor, 2021-11-16:
@bpaser @mohneesh7 Can you test emlearn 0.13.2 ? I believe the issue might be fixed there.

jonnor, 2022-11-23:
Has anyone been able to test this?

jonnor, 2023-07-11:
Tested this now on Window 10, with emlearn 0.17.0. Works fine. Probably has since 0.13.1. Closing

closed 2023-07-11 by jonnor