emlearn/emlearn #41: Regression Support

mohneesh7, 2021-10-05:
Basically the title, Is there any regression support, because All I see is classifiers, If not, is there any plan to implement it soon.

jonnor, 2021-10-05:
Right now there is no regression support. Are there any particular models you have in mind for regression? Regression trees (RandomForest etc) is tracked in #31

mohneesh7, 2021-10-06:
Hello, I have tried using a keras sequential model as a regressor, Although I think it worked, I am having trouble running the c code to use the generated header file, even after pasting the eml_net.h and eml_common.h files.
Apart from that, basic regressions like linear and isotonic would be nice to work on

jonnor, 2023-04-16:
Regression with neural networks might kind-of-work, as they already output a continuous value.
But this should be tested. Both in unit-tests, and adding it to our regression example (https://emlearn.readthedocs.io/en/latest/auto_examples/regression.html#sphx-glr-auto-examples-regression-py).

Makes sense to support basic linear regression also.

adam-ah, 2023-07-16:
Late to the party, but this is a dead simple fix. 
Currently, the generated code calls `int32_t eml_net_predict`, which binarizes the data with `_class = (model->activations2[0] > 0.5f) ? 1 : 0;`
Simply returning `model->activations2` is the solution.

Add this to `eml_net.h`:
```
float_t*
eml_net_regress(EmlNet* model, const float* features, int32_t features_length) {

    const EmlError error = eml_net_infer(model, features, features_length);
    if (error != EmlOk) {
        return -error;
    }

    const int32_t n_outputs = eml_net_outputs(model);

    return model->activations2;
}
```

and this to the generated `.h` file:
```
    float_t*
    whatevername_model_regress(const float* features, int32_t n_features)
    {
        return eml_net_regress(&keras_model, features, n_features);
    }
```

jonnor, 2023-07-16:
Hi @adam-ah - thank you for the feedback.

Agreed that the everything-is-classifier is bad. We are moving away from that. In fact, over time I am considering removing the APIs that do the thresholding/argmax completely, to force people to make a decision themselves.

We are working to standardize on 

So the preferred API would be like this, even if it means we need a copy:
```
EmlError eml_net_regress(EmlNet* model, const float* features, int32_t features_length, float *out, int32_t out_length)
```
And then we can have a convenience for the common case where there is only a singled-value output.
```EmlError eml_net_regress1(EmlNet* model, const float* features)```
Ref how `eml_trees_regress1` is implemented.

Would you be interested in making a merge request to integrate this functionality? 

In addition to the C code implementation, it also needs:
- Outputting the whatevername_model_regress convenience wrapper function. In emlearn/net.py c_generate_net - can see trees.py for an example
- Supporting the regression codepath in the Python wrapper. See Wrapper in emlearn/net.py. Can see trees.py for example code
- A unit test. Like we have for classification: https://github.com/emlearn/emlearn/blob/master/test/test_net.py

adam-ah, 2023-07-17:
I'm more than happy to contribute more than that, @jonnor as long as the `citation` is not for *Jon Nordby* but for all contributors.

jonnor, 2023-07-17:
That makes sense! I agree with that :)

jonnor, 2023-08-08:
Support for Keras Sequential for regression is now complete, thanks to @adam-ah. Closing this issue.

Support for Linear regression is tracked in #56  and for scikit-learn MLP regressors in #63

closed 2023-08-08 by jonnor