emlearn/emlearn #42: An example C code that outputs the prediction

mohneesh7, 2021-10-06:
Hello, I have created a model.h header file, but unable to import that and make a prediction in a C program. Anyone who has done this can you assist?

jonnor, 2021-10-06:
Where are you stuck?
The README file has a basic example of the workflow. But you are right that there should be a complete example that can be ran end-to-end by just following a couple of steps.

mohneesh7, 2021-10-07:
Yeah If I succesfully run the C code then maybe I will try a pull request with an example. I am trying to convert a Keras sequential model into C program implementation, The generated .h file needs eml_net.h file and eml_common.h file as well, after that as well, I am not able to call the eml_net_predict function in the C code.
A clear example would be very nice.

jonnor, 2021-12-28:
There is now a complete example of classification in examples/xor.py / examples/xor.c
It is built as part of the docs and can be seen here: 
https://emlearn.readthedocs.io/en/latest/auto_examples/xor.html#sphx-glr-auto-examples-xor-py

closed 2021-12-28 by jonnor