emlearn/emlearn #59: Mixture model don't support export to c file

cnxdeveloper, 2023-05-16:
Thank you for your awesome repo. I tried to apply Gaussian Mixture from Sklearn to emlearn and I have tried to convert emlearn to C code file. But I got the problem. It's raised "raise NotImplementedError("TODO implement save()")". Pls help me to solve it

jonnor, 2023-05-23:
This should now work in git master. Example code below

```python
# gmm_iris.py

from sklearn import datasets
from sklearn.mixture import GaussianMixture
import emlearn

iris = datasets.load_iris()
estimator = GaussianMixture(n_components=3, covariance_type='tied', max_iter=20, random_state=0)
estimator.fit(iris.data)

print(iris.data[iris.target == 0].mean(axis=0))

save_path = 'out_gmm.c'
cmodel = emlearn.convert(estimator)
cmodel.save(file=save_path, name='my_gmm')
print('wrote', save_path)
```

```C
// gmm_iris.c

#include "out_gmm.c"
#include <stdio.h> // printf
#include <stdlib.h> // stdod

#define N_FEATURES 4
#define N_COMPONENTS 3
static float features[N_FEATURES] = {0.0};
static float output[N_COMPONENTS] = {0.0};

int
main(int argc, const char *argv[])
{{
    const int n_features = N_FEATURES;
    const int n_components = N_COMPONENTS;

    if (argc != 1+n_features) {{
        fprintf(stderr, "Wrong number of arguments\n");
        return -1;
    }}

    for (int i=1; i<argc; i++) {{
        features[i-1] = strtod(argv[i], NULL);
    }}

    float score;
    const EmlError out = my_gmm_score(features, n_features, output, &score);
    if (out != EmlOk) {{
        return -out; // error
    }}

    printf("score=%f\n", score);
    for (int i=0; i<n_components; i++) {
        const float prob = exp2f(output[i]);
        printf("c=%d log_prob=%.2f prob=%.2f\n", i, output[i], prob);
    }

    return 0;
}}
```

Running these commands. Assuming that GCC is installed. And using bash or similar as the shell.

```shell
python gmm_iris.py
gcc -o gmm -I$(python -c 'import emlearn; print(emlearn.includedir)') -lm gmm_main.c && ./gmm 5.0 3.4 1.5 0.5
```

Should output
```
wrote out_gmm.c

score=-1.109145
c=0 log_prob=-1.11 prob=0.46
c=1 log_prob=-71.94 prob=0.00
c=2 log_prob=-34.68 prob=0.00

```

closed 2023-05-23 by jonnor

jonnor, 2023-05-23:
Released in emlearn `0.16.5`

PR #60 (2023-05-23): Fix save() for GaussianMixtureModel
Fixes #59
commit 7b91030f (2023-05-23): mixture: Fix save() not implemented for GaussianMixtureModel

Fixes #59
Also removes one hardcoding of output directory