emlearn/emlearn #70: River x emlearn

MaxHalford, 2023-10-19:
Hey there! I'm a core maintainer of [River](https://github.com/online-ml/river/), a Python library for online machine learning. Do you have a stance on online learning? I always thought it would be cool to not only do inference on microcontrollers, but also learning.

jonnor, 2023-10-19:
Hi there @MaxHalford - thanks for reaching out. river is nice work, kudos for that! I have tested it briefly a while back, and one of our collaborators at Soundsensing built https://github.com/NorskRegnesentral/streamchange which is inspired by river. 

I am generally open to having learning algorithms also in emlearn, provided that they are suitable for microcontrollers. Meaning that the algorithms are useful for the kind of tasks microcontrollers are used for, and that decent performing models can be done within the computational constraints of a microcontroller (RAM, FLASH and CPU). And kind-of like scikit-learn, we prefer more established/proven methods.

At least in theory, the online approach to learning should be beneficial in the microcontroller setting, as it avoids needing to keep an entire dataset in memory. But I must admit that I have not yet looked into the suitability of the existing algorithms. Do you have any particular usecases and/or algorithms in mind?

Personally, I am interested in a few areas where continious learning

* Anomaly Detection, especially for Condition Monitoring of machines.
* Adaptive sampling. Reducing how often sensors need to be turned on is a major enabler of power saving, and thus battery operation, of IoT sensor devices. Having an online model-based estimation of the data-generating process.

jonnor, 2023-10-19:
My goal is to include some more preprocessing in emlearn. These would primarily be streaming-based, to keep memory usage low. So that is another area which is quite similar to river.

MaxHalford, 2023-10-20:
Hey there @jonnor, thanks for the context!

> I am generally open to having learning algorithms also in emlearn, provided that they are suitable for microcontrollers. Meaning that the algorithms are useful for the kind of tasks microcontrollers [...]

I completely understand. We've recently starting working on a new project called [LightRiver](https://github.com/online-ml/light-river/tree/ahnentafel): we take the best algorithms in River and implement them in Rust. We're actually starting with anomaly detection, because it's a common use-case for streaming data. Our belief is that half-space trees (HST) is one of the best algorithms here: it uses a forest of random trees to approximate the data distribution. It works really well as it is robust to drift (when the distribution gradually/abruptly changes).

And yes I agree that KNNs are a good candidate too.

I'm not sure how useful LightRiver is to you if it's written in Rust, but I'm open to making something work :)

jonnor, 2023-10-20:
Half-space trees sounds relevant. It could probably be built on top of our existing tree-based inference code. I would welcome a contribution for that. A related task is support for IsolationForest #34 

Considering that we would need to implement any learning algorithm that is to run on-device in microcontroller-friendly C, I think that River/LightRiver will be most useful as a reference implementation. Today we test our (inference) implementations against scikit-learn/Keras, and any online learning implementation could/should probably be tested against River.