emlearn/emlearn #75: How to fixed point?

adur1990, 2023-11-17:
Hi,

I am currently evaluating ML inference frameworks for microcontrollers with no glibc and no FPU. Therefore, I am more or less forced to use a framework that supports fixedpoint/integer models. emlearn states that it is supported but there is no documentation. How would I continue if I wanted to re-implement your XOR example using fixed point/integer mode?

Thanks :)

jonnor, 2023-11-18:
Hi @adur1990 - thank you for the question. You are right, this is not properly documented at the moment.

This feature is supported for tree-based models (RandomForest etc).
To use this feature, make sure all the input data for training/prediction is scaled to be integers that fits in 8/16/32 bits,
and set the `dtype` argument of `emlearn.convert()` to the appropriate C datatype. For example: `emlearn.convert(model, dtype='int8')`.

jonnor, 2023-11-18:
I am adding this to the docs, and also working on an example of its effect on model/program memory size for non-FPU platforms. This happening in #76

adur1990, 2023-11-19:
Thank you for the quick response. It's a bummer that it does not work für neural nets... but okay, let's see how far I will get with this. Do you have a pointer how I make sure that the scikit-learn RandomForestClassifier is scaled to integers? Thanks :)

jonnor, 2023-11-19:
One simple approach is to use scikit-learn MinMaxScaler() to map the features in dataset to 0.0, 1.0 range - and then map this to integers. For example with int16, one can use `(MinMaxScaler().fit_transform(X) * 32768).astype(int)`

jonnor, 2023-11-19:
Fixed-point support for neural networks is in progress, I have created a separate issue for that.
There is now documentation with examples regarding integer computations with tree-based models.
https://emlearn.readthedocs.io/en/latest/auto_examples/trees_feature_quantization.html

closed 2023-11-19 by jonnor

adur1990, 2023-11-23:
I have to add another comment here, sorry. Unfortunately, the learned estimator is still using floating point. I want to really get rid of all floats as my target platform has no FPU. Is this a bug or do i miss something? Here is the code I'm trying to get running:

```python
import emlearn
import numpy
from sklearn.ensemble import RandomForestClassifier
from sklearn.preprocessing import MinMaxScaler

# Generate simple dataset
def make_xor(lower=0.0, upper=1.0, threshold=0.5, samples=100, seed=42):
    rng = numpy.random.RandomState(seed)
    X = rng.uniform(lower, upper, size=(samples, 2))
    y = numpy.logical_xor(X[:, 0] > threshold, X[:, 1] > threshold)

    return X, y

X, y = make_xor()
X = (MinMaxScaler().fit_transform(X) * 127).astype(int)

estimator = RandomForestClassifier(n_estimators=10)
estimator.fit(X, y)

# Convert model using emlearn
path = 'xor_model.h'
cmodel = emlearn.convert(estimator, method='inline', dtype='int8_t')
cmodel.save(file=path, name='xor_model')
```

jonnor, 2023-11-24:
Ah. The selection of data type to use is only respected for the inline inference strategy. And by default, the save() call will output code for both the "loadable" strategy (floats) and inline. I just released 0.19.1 where this is configurable. Add inference=['inline'] to the save() call to avoid the float code being there.

adur1990, 2023-11-27:
Something is still off... using the inline method the model is not exported at all... I attached two version of the generated header file, the first one using the code from above and the second one changing

```
cmodel.save(file=path, name='xor_model')
```

to

```
cmodel.save(file=path, name='xor_model', inference=["inline"])
```

Furthermore, the inference is still using floating points.

Thank you again for your support.

[xor_models.zip](https://github.com/emlearn/emlearn/files/13472229/xor_models.zip)

jonnor, 2023-11-27:
The loadable model not being exported is expected, that is exactly what the option is for.
But the existence of floating point literal is a bug. I was able to reproduce it. Fix in progress at https://github.com/emlearn/emlearn/pull/79

Btw: which platform are you targeting? Does it have support in an open compiler, that we could run in CI? That could help ensure these things continue to work correctly. Many platforms just inject soft-float automatically support when needed, so these issues sometimes stay hidden.

adur1990, 2023-11-30:
So I'm aiming for a somewhat niche/esoteric target. I'm trying to use the [Nexmon platform](https://nexmon.org) to perform ML on the Wi-Fi chip of a smartphone, in my case the Pixel 7. Nexmon uses a more or less normal GCC, which also injects soft floats. But since my use case is somewhat performance dependent, I want to avoid soft floats.

jonnor, 2024-01-03:
The floating point literal issue was fixed in 0.19.3, which was just released.

adur1990, 2024-01-08:
Thank you, that issue seems to be fixed now.

jonnor, 2024-01-08:
@adur1990  Thank you for testing and confirming! Anything else, then just open a new issue :)