emlearn/emlearn #77: Integer/fixed-point support for neural networks

jonnor, 2023-11-19:
Right now trees, bayes and neighbors all have some support for integer/fixed-point only inference. However nets (multi layer perceptron) does not.

There are many platforms without floating-point support. And neural networks are the most performant for several kinds of tasks. So it would be great to have support for integer-based computations there also.

jonnor, 2023-11-19:
There has been prep work for this done in #74 - but it is not functional yet. Most of the infrastructure should be in place, but the core computations are missing.

jonnor, 2023-11-19:
This was requested by @adur190 in #75.

And others have expressed interest/need for this before. CC @irenebru and @jorgegm10

adur1990, 2023-11-21:
Thanks for adding this to your enhancements list :)

ss22ongithub, 2025-06-06:
Dear @jonnor 
I am interested in performing time-series based forecasting using emlearn  on  a device that lacks FPU . 
Is there a documentation or any pointers that you could suggest?
Thanks