emlearn/emlearn #80: Overflow warnings when converting RandomForestsClassifier

mrkajetanp, 2023-12-13:
Hi,

Apologies if I'm just missing something obvious but when trying to convert a RandomForestsClassifier (trained on all-integer data) I keep getting tons of these warnings:
```
In file included from tmp/myinlinetree.c:2:
tmp/myinlinetree.h:62312:23: warning: overflow in conversion from ‘int’ to ‘short int’ changes value from ‘66413’ to ‘877’ [-Woverflow]
62312 |   { 6, 1736.5, 62303, 66413 },
      |                       ^~~~~
tmp/myinlinetree.h:65535:23: warning: overflow in conversion from ‘int’ to ‘short int’ changes value from ‘66017’ to ‘481’ [-Woverflow]
65535 |   { 4, 6592.5, 65526, 66017 },
      |                       ^~~~~
tmp/myinlinetree.h:65536:23: warning: overflow in conversion from ‘int’ to ‘short int’ changes value from ‘65809’ to ‘273’ [-Woverflow]
65536 |   { 3, 4978.5, 65527, 65809 },
      |                       ^~~~~
```
Any idea how I could fix that?
I was using this line to convert it, both with and without the dtype argument - same effect.
```
cmodel = emlearn.convert(forest, method='inline', dtype='int32')
```

jonnor, 2024-01-02:
Hi @mrkajetanp - thank you for reporting. Which platform and compiler is this?

jonnor, 2024-01-02:
How large is your RandomForest? I believe you might be running out of space to store the leaves - only 16 bits are used, giving max 32767 nodes total (around 282 kB). Try reducing the number of trees and/or limit the depth of the trees?

Running out of nodes should probably be an error (or at the very least a warning) in Python when doing the conversion. Instead of it happening only at compile time.

jonnor, 2024-01-02:
It was pointed out to me by @mastensg that the leaf values could probably be relative jumps instead of absolute as they are now. That would enable much larger models for the same model size.

mrkajetanp, 2024-01-03:
> Which platform and compiler is this?

This was on Gentoo Linux with whichever compiler is used by default so presumably gcc?

>  I believe you might be running out of space to store the leaves

Ah yes that'll most likely be it then. Is there any way I could check the exact number of nodes I have in the tree? I know the joblib serialization files for it are much bigger than 282 kB but those two wouldn't exactly be comparable.

> the leaf values could probably be relative jumps

It sounds like that could work and it would be great to support these larger model sizes.  Do you think you'd be able to implement that in the near future? I'd love to use the functionality for some academic research I'm doing at the moment but probably wouldn't have time to dig in and do it myself :/

jonnor, 2024-01-03:
Gentoo, then I am assuming it is x86_64. But yeah on further thought, it probably is not relevant here.

The number of nodes can be returned using [emlearn.evaluate.trees.model_size_nodes](https://emlearn.readthedocs.io/en/latest/evaluate.html#emlearn.evaluate.trees.model_size_nodes). It takes as input a scikit-learn model.

I cannot promise that I will be able to implement support for larger models in the short term. But I might get a bit more time towards the end of January. A quick hack that might be acceptable for you, being on a not-so-constrained platform, would be to increase the size of datatype used for the leaf nodes.

jonnor, 2024-01-03:
I quickly checked if relative jumps could be implemented easily. It would require more work unfortunately, as documented in #81

So for now we are limited to max 32768 nodes in a decision tree ensemble.
A model with 32768 will currently take nearly 400 kB (12 bytes per node), which is quite a lot for microcontrollers (our primary target).  A fully grown tree with max_depth=10 takes 2048 nodes, so with such depth n_ensembles would be maximum of 15.
If using max_depth=8 is 512 nodes, it coudl to support 60 fully grown trees.
When using trees that are not fully grown, the number of trees / maximum depth can be somewhat larger than this.

So one should limit n_ensembles and tree depth somehow to constrain model size. There are examples of this in: 
https://emlearn.readthedocs.io/en/latest/auto_examples/trees_hyperparameters.html

jonnor, 2024-01-07:
Might be possible to give this another chance now, with #62 being merged

jonnor, 2024-01-07:
This should be fixed now in git master.

@mrkajetanp can you test if it fixes your issue?

It can be installed using
```
pip install git+https://github.com/emlearn/emlearn.git
```

closed 2024-01-07 by jonnor

mrkajetanp, 2024-01-09:
Oh that's nice, much appreciated! How many nodes could the new implementation accommodate?

jonnor, 2024-01-09:
@mrkajetanp theoretically it is now 2**15 per tree, and up to 2**31 decision nodes total. So up to 2**15 trees, if they are 2**15 each. I have added a test that checks 1000 trees and over 100k nodes. More than that was very slow to construct.

mrkajetanp, 2024-01-14:
Thank you, I checked and the implementation works well now for any size of tree I tried, much appreciated!

All I'm getting are these warnings, is that to be expected?
```
In file included from tmp/myinlinetree.c:2:
tmp/myinlinetree.h:4341:9: warning: initialization discards ‘const’ qualifier from pointer target type [-Wdiscarded-qualifiers]
 4341 |         myinlinetree_leaves,
      |         ^~~~~~~~~~~~~~~~~~~
tmp/myinlinetree.c: In function ‘classify’:
tmp/myinlinetree.c:6:64: warning: passing argument 1 of ‘myinlinetree_predict’ from incompatible pointer type [-Wincompatible-pointer-types]
    6 |             printf("%d,%f\n", row, (float)myinlinetree_predict(values, length));
      |                                                                ^~~~~~
      |                                                                |
      |                                                                const float *
tmp/myinlinetree.h:21648:45: note: expected ‘const int32_t *’ {aka ‘const int *’} but argument is of type ‘const float *’
21648 | int32_t myinlinetree_predict(const int32_t *features, int32_t features_length) {
      |                              ~~~~~~~~~~~~~~~^~~~~~~~
In file included from tmp/myinlinetree_proba.c:2:
tmp/myinlinetree_proba.h:4341:9: warning: initialization discards ‘const’ qualifier from pointer target type [-Wdiscarded-qualifiers]
 4341 |         myinlinetree_leaves,
      |         ^~~~~~~~~~~~~~~~~~~

```
The model was trained on a dataframe where every column was an int32.

jonnor, 2024-01-16:
Thank you for testing and verifying the fix. The first warning there (about leaves) I also spotted and have fixed in master, to be released soon. It did not cause any particular problem though.

The second warning *could* potentially cause problems when running .predict() from Python - but is probably also inconsequential (it happens during the tests which do check correctness, at least on host/x86_64). But I will try to fix it, and hopefully also enable -Werror for the tests so that we detect if any new (and potentially more serious) warnings show up.

jonnor, 2024-01-16:
I filed a separate issue regarding the warnings, https://github.com/emlearn/emlearn/issues/87

PR #82 (2024-01-07): Trees: Make decision node jumps relative
Should fix #80