emlearn/emlearn #98: Dedicated Arduino Library

Ryan-py, 2024-06-04:
I have used emlearn for a few project and I really like it , Is it possible to have a dedicated Arduino library , this will open emlearn to more users and also make the installation process much easier . 

I am also happy to help in developing this.

jonnor, 2024-06-09:
Hi @Ryan-py. Glad to hear that you have found the project to be useful!

Have you seen the latest documentation for emlearn on Arduino? Specifically the "Install emlearn as Arduino library" section? There is now a single command which creates and installs an Arduino library.  
https://emlearn.readthedocs.io/en/latest/getting_started_arduino.html#install-emlearn-as-arduino-library

jonnor, 2024-07-07:
Closing, since there is a fix for this given in the previous comment.

closed 2024-07-07 by jonnor