emlearn/emlearn #102: Is it possible to write a customized inference function within emlearn

DylanWangWQF, 2024-08-23:
Hi, I just read your 'getting started' tutorials. 
Most of the functions like `training()` and `predict()`  are implemented  using the `sklearn`.
So, If it's possible to train a decision tree model within `emlearn`, save the model locally, and then implement the customized inference function which will be used for the following inference?
For example, I want to change the input feature values in one of the tree nodes during the inference. 
Basically, I need to control the process of the inference flexibly.

jonnor, 2024-08-24:
Hi @DylanWangWQF. There is no specific support or hooks for anything like that in emlearn. And I do not think that I would add it either, unless there would be a very wide-spread usecase for it.
But as emlearn is an open source project, you are of course to take the code and modify it as you see fit for your particular needs. Or just to take aspects that you like, for example how it reads the node information from scikit-learn or how the C code generation works.

DylanWangWQF, 2024-08-24:
@jonnor Thanks for your suggestions!

closed 2024-08-24 by DylanWangWQF

jonnor, 2024-08-24:
You are welcome! Best of luck with your project. Do send me a link if you get something working (with or without emlearn), as I love research into custom inference for tree-based models!

DylanWangWQF, 2024-08-24:
> You are welcome! Best of luck with your project. Do send me a link if you get something working (with or without emlearn), as I love research into custom inference for tree-based models!

Sure, I will!