emlearn/emlearn #103: Converted model file 'tree.h' does not match the trained tree

DylanWangWQF, 2024-08-28:
Hi @jonnor 
As mentioned in #102 , I'm going to write my specific`inference` based on `emlearn`.
Before start, I wrote a simple test according to `examples/classifiers.py`.
I tested the accuracy using the trained model from `sklearn` and the converted model from `dt_model.h` and found that the accuracy is different due to the different trees.
```
// code for converting the model
path = 'micro/dt_model.h'
cmodel = emlearn.convert(clf, method='inline', dtype='float')  # loadable, inline
cmodel.save(file=path, name='dt_model')

// dt_model.h
if (features[10] < 9.075000f) {
    return 3;
} else {
    return 2;
}
...
```
This  [tree pdf](https://github.com/user-attachments/files/16785867/dt.pdf) is a plotted tree after training with `sklearn`, you can see that most of the labels are different. But the threshold values are correct.

Additionally, I set `dtype='float'` in `convert()`, otherwise the threshold values in the converted tree would be approximated to integers. However, after using `float`, it shows that there are many `implicit conversion from 'float' to 'int16_t'` in `EmlTreesNode` object. Not sure if it will impact the comparisons.

Here is the full code ([inf.txt](https://github.com/user-attachments/files/16785979/inf.txt)) and converted tree `.h` ([dt_model.h.txt](https://github.com/user-attachments/files/16785978/dt_model.h.txt))

Basically, I don't know if this is a bug or I missed something?
BTW, the `convert` function is really a good option for me since it allows me to flexibly modify the threshold values.
Thanks!

DylanWangWQF, 2024-09-02:
Additionally, in `dt_model.h`, we have `int32_t votes[6] = {0,};`. However, as mentioned, the class in the trained tree could be `6/7`. `votes[7]` is not expected. Did I miss something in the`convert`?

jonnor, 2024-09-04:
When using a depth limiter like max_depth, there may be differences to scikit-learn, because in git master, emlearn uses hard majority voting (top1 class of leaf) compared to scikit-learn which uses soft majority voting (mean of class proportions in leaf). This is discussed in https://github.com/emlearn/emlearn/issues/12

jonnor, 2024-09-04:
The `implicit conversion from 'float' to 'int16_t'` with EmlTreesNode getting float data when the nodes are expected to be integers. This is because we switched to integer based inference for the "loadable" stategy in the latest release. And because we default to outputting both inference strategies in save(). It may be removed by passing inference=['inline'] to save().

jonnor, 2024-09-04:
> Additionally, in dt_model.h, we have int32_t votes[6] = {0,};. However, as mentioned, the class in the trained tree could be 6/7. votes[7] is not expected. Did I miss something in the convert?

I do not understand what you mean by "the class in the trained tree could be 6/7. votes[7] is not expected". Can you clarify?

DylanWangWQF, 2024-09-05:
> I do not understand what you mean by "the class in the trained tree could be 6/7. votes[7] is not expected". Can you clarify?

```
typedef struct _EmlTrees {
    int32_t n_nodes;
    EmlTreesNode *nodes;

    int32_t n_trees;
    int32_t *tree_roots;

    int32_t n_leaves;
    uint8_t *leaves;
    int8_t leaf_bits;

    int8_t n_features;
    int8_t n_classes;
} EmlTrees;
```
Sorry for the unclear problem. I misunderstood something. `n_classes=6` and `int32_t votes[6] = {0,}` mean that we will calculate the votes for the six possible classes (which I mistakenly thought was for leaves)

DylanWangWQF, 2024-09-19:
Thanks. Yes, this has been resolved.