emlearn/emlearn #108: Practical examples using sensor data for Zephyr

jonnor, 2025-03-11:
Right now the emlearn module for Zephyr only has a "hello world" example, which just runs on data that is in-memory.
This is useful to get started with using the C code. But does not help much in connecting up machine learning modules to sensor data (which is the primary usecase).

So it would be very useful with a more complete example code, which shows how to use one of the Zephyr sensor APIs together with emlearn. Relevant hardware could for example be IMU/accelerometer, using the "sensor" API. Right now most drivers only support "Fetch and Get" style API (individual datapoints), though there is move to a "Read and Decode" API which is generally more suited for continious-sampling.

jonnor, 2025-05-08:
I think a key part of this will be some code that does continuous reading of IMU using Sensor API and collecting into the desired. 
There is a "magic wand" example in Zephyr already that processes IMU data - however, that does _not_ seem to do continuous readout/classification - it seems that sensor readout is not done when ML processing/inference is going on - so data samples are lost.

To avoid missing data that, I believe the sensor readout should happen using its own thread (with semi-high priority). And the thread should push ready-to-consume windows of data onto a queue. This way, the application in main can check this queue occasionally and process the data at a leisurely pace. Including doing feature extraction and classifier (using emlearn).

The most suitable sensor API in Zephyr is the new "decode API". However, a bunch of drivers do not yet support this, so it would be good to also have compatibility with the older "fetch & get" API.

jonnor, 2025-05-08:
Pseudocode for an application using the kind of "sensor reader" helper described above.

```C

// Configuration
#define SAMPLERATE 100
#define WINDOW_LENGTH 100
#define HOP_LENGTH 25

#define N_CHANNELS = 3;
enum sensor_channel sensor_reader_channels[N_CHANNELS] = {
	SENSOR_CHAN_ACCEL_X,
	SENSOR_CHAN_ACCEL_Y,
	SENSOR_CHAN_ACCEL_Z,
};

// Reader internals
sensor_value sensor_reader_output_buffer[WINDOW_LENGTH*N_CHANNELS];
sensor_value sensor_reader_new_buffer[HOP_LENGTH*N_CHANNELS];

K_THREAD_STACK_DEFINE(sensor_reader_stack, MY_STACK_SIZE);
struct k_thread sensor_reader_thread;

K_MSGQ_DEFINE(sensor_reader_queue, sizeof(struct data_item_type), 1, 1);

struct sensor_reader {
    .samplerate=SAMPLERATE,
    .thread=&sensor_reader_thread,

    .read_samples=sensor_reader_new_buffer,
    .read_samples_index=0,

    .output_buffer=sensor_reader_output_buffer,
    .output_buffer_index=0,

    .n_channels=N_CHANNELS,
    .channels=sensor_reader_channels,

    .dev=sensor,
    .queue=&sensor_reader_queue,
    .thread=&sensor_reader_thread,
}

static int set_sampling_freq(const struct device *dev, int samplerate)
{
	int ret;
	struct sensor_value odr;
	odr.val1 = samplerate;
	odr.val2 = 0;
	ret = sensor_attr_set(dev, SENSOR_CHAN_ACCEL_XYZ, SENSOR_ATTR_SAMPLING_FREQUENCY, &odr);

	return 0;
}

int main(void) {

    struct sensor_chunk_msg chunk;

    if (!device_is_ready(sensor) {
        printk("sensor: device %s not ready.\n", sensors[i]->name);
        return 0;
    }

    // Setup sensor
    set_sampling_freq(sensor, SAMPLINGRATE);

    // Start high-priority thread for collecting data
    sensor_reader_start(&sensor_reader);

    while (1) {

        // check for new data
        const int get_status = k_msgq_get(sensor_reader.queue, &chunk, K_NO_WAIT);
        if (get_status != 0) {
            // do the processing
            process(chunk.buffer, chunk.length);
        }

	    k_msleep(100);
    }

    return 0;
}
```

jonnor, 2025-05-08:
A potential demo application would be "walking detection". It is easy to test. The model proposed in [this paper](https://www.nature.com/articles/s41746-022-00745-z ) is designed to be "universal" (work for a range of sensor placements), tested on a bunch of datasets, and is conceptually quite simple. I believe it is possible to replace the wavelet transform with an IIR filter or FFT (both already supported in emlearn).

kartben, 2025-08-24:
> There is now some basically-functional code for the thread+queue sensor readout, and feature extraction for accelerometer (for Human Activity Recognition etc) at [jonnor/zephyr@`emlearn-sensor-readout`/samples/modules/emlearn/sensor_reader](https://github.com/jonnor/zephyr/tree/emlearn-sensor-readout/samples/modules/emlearn/sensor_reader?rgh-link-date=2025-08-24T21%3A56%3A56.000Z)

I sense (pun intended) a Zephyr Developer Summit talk in your near future :)

jonnor, 2025-08-24:
> I sense (pun intended) a Zephyr Developer Summit talk in your near future :)

Hah, it is rather imminent :p - these examples became a bit more JIT than originally planned... See you tomorrow :)