emlearn/emlearn #112: Random Forest binary classifier predict_proba method

codewarrior26, 2025-06-02:
Thank you for this wonderful repo!

I'm currently using emlearn to run a Random Forest classifier. I noticed that the generated code uses `if-else` conditions and `return <class>` statements to perform hard classification.

I wanted to ask:
- Does emlearn support the equivalent of '**predict_proba(**)' from scikit-learn?
- If not directly, is there a recommended way to retrieve class probabilities for binary classification?

jonnor, 2025-06-30:
Hi @codewarrior26. The "loadable" inference mode for trees supports returning probabilities. Then you would use the eml_trees_predict_proba() function.

Each tree still does hard voting, with a single class as the outcome. But with say 10 trees (or more) this is often enough to be a useful continuous score.

jonnor, 2025-07-14:
With https://github.com/emlearn/emlearn/pull/116 there are wrapper functions provided for mymodel_predict_proba() (where mymodel is the specified name), both for inline and loadable case.

jonnor, 2025-07-29:
Above work to support class proportions has now landed. I have also added mention of predict_proba() on the documentation for tree-based models.